metalman3797/Cura-Dremel-Printer-Plugin #1: plugins/DremelOutputDevice folder

WeavingColors, 2017-11-16:
Hello! 

"plugins/DremelOutputDevice" folder in the zip folder does not exist in the zip file download posted on this
 
I am encountering an error when trying to save the Dremel file in Cura. 
The error states that the /DremelOutputDevice does not exist.

Thank you so much for your assistance and guidance.

timmehtimmeh, 2017-11-16:
Hi - sorry, there was an error in the documentation that's been fixed now.  you should copy plugins/DremelGCodeWriter to the plugins folder of your cura install

WeavingColors, 2017-11-16:
That is exactly what I thought the issue was. 
Luckily, I assumed that was the issue and tried that out.
Unfortunately, I am still experiencing an error when saving the file. 

The Dremel 3D20 just doesn't want to budge. Im not sure what to do. 
I followed all of the directions you have posted and it seems to work correctly.
It just wont print. 

(I work for the Pima County Public Library in the 101 Space and we are trying to use Cura to print our files...Thank you sooo much from PCPL!)

WeavingColors, 2017-11-16:
I will attempt to screen capture the error screen that pops up in Cura and post it to this thread.

timmehtimmeh, 2017-11-16:
@WeavingColors  - thanks for helping debug this - I'll look into the issue tonight.

WeavingColors, 2017-11-16:
Also, the Cura logo does not pop up as the preview on the Dremel. It is just blank. 

@timmehtimmeh You are most certainly welcome! Thank you so much for making this a reality. We appreciate your assistance and guidance.

timmehtimmeh, 2017-11-16:
@WeavingColors  - when you post screenshots, could you also see if you could post the .g3drem file that was generated...thanks

WeavingColors, 2017-11-16:
@timmehtimmeh Yeah! Will do. I will post ASAP.

timmehtimmeh, 2017-11-16:
@WeavingColors - also if you could let me know which OS and version of cura you're using that would help greatly

WeavingColors, 2017-11-16:
[image]
[image]
[image]

WeavingColors, 2017-11-16:
[Dremel File.zip](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/files/1480434/Dremel.File.zip)

WeavingColors, 2017-11-16:
I believe that is everything you requested. Please let me know if I can provide additional information.

timmehtimmeh, 2017-11-16:
@WeavingColors - awesome...thanks for the info.  As suspected the plugin wasn't finding the cura bitmap.  I've moved the bitmap data inside the plugin, so download the zip file again, and overwrite plugins/DremelGCodeWriter/DremelGCodeWriter.py with the new one.  Let me know if the new version works for you.

WeavingColors, 2017-11-16:
Absolutely lovely! The printer is up and running now!

Thank you so much from me, PCPL, and the 101 Space staff.
You have been absolutely terrific!

Best wishes!

timmehtimmeh, 2017-11-16:
Awesome to hear. Thanks for your help as well.  Next time I'm in Tucson maybe I'll swing by to check out your setup.

closed 2017-11-16 by timmehtimmeh