metalman3797/Cura-Dremel-Printer-Plugin #5: Linux (Ubuntu 17.10) Cosmeticproblems

SwapFaceL, 2018-02-19:
Great work. It works like a charm, slicing and printing is not a problem at all.

Just two points:

1) The Gramm Prediction is not accurate as well as the printing time preditction. I adjusted the Material settings to 1,07g/cm³ which corrects the weight prediction.
Any way to also correct the time prediction without change the produced G-Code, because the Prints are totally fine.

2) Screenshot - The implementation for the Screenshot is not working on my system. I just see black squares as preview. Can't even see the red box. 
Maybe you can have a look and make it Compatible so I can enjoy nice Preview Pictures as well :)?

timmehtimmeh, 2018-02-19:
Hi @SwapFaceL  -  thanks for testing this on Linux and reporting back.  I'm very happy that I'm able to help users like you make the printer work with operating systems that are not officially supported by Dremel! Currently I don't have a linux machine, but I'll work on setting up Ubuntu on a pen drive and try testing that out.   Which version of cura and which version of the plugin are you using?

Regarding your first point I've got a quick question:  did you adjust the density in grams/cm³ on the dremel_pla material or another material, and are you using the official dremel material while doing the prints?  I'm more than happy to change the set density in this repo, as I haven't strictly measuered dremel's PLA format (I've long since switched to 3rd party filaments), but I got the spec from dremel's own officially released "Dremel DigiLab 3D Slicer" that uses cura underneath it.  Maybe they didn't set the filament density up correctly, but I'd be a bit surprised by that.  It's also possible that there is an incorrect estimate on Cura's end on the amount of material extruded that would cause an incorrectly prediced weight.

Regarding how to change the estimated time:  Unfortunately no I have no control over that - I get that number straight from Cura (along with the estimated filament length).  I've found that even Simplify3D, which is a paid product can produce significant errors in the time estimate, often on longer prints S3D will underestimate the actual amount of time required by hours.

I'll take a look into the screenshot issue you're seeing in Linux.  Inside the plugin there's some fallback code that will write a generic Cura icon instead of the screenshot if the screenshot section detects some errors, but apparently the issue you're seeing isnt' triggering that code.

SwapFaceL, 2018-02-20:
Hi,

wow a lot to process now, so I will try to answer as structered as possible.
**System-Configuration**
Ubuntu 17.10
Cura 3.2.0
Dremel 3D20 (obvious)
with original dremel_pla

**Prediction-Question**
As you already found out: I changed the density of the original dremel_pla (you provided). I didn't care about the filament details Dremel provided. I just checked the weight of some of my printed objectes and adjusted the value until it was a correct weight prediction in all cases. **So the density I instered might be not the correct one, but helps Cura to predict the weight correctly**.
The Calculation of Curas time prediction is a pity, would be nice if we could adjust it, because I think it's just a linear factor which is missing.

**Screenshot Issue**
Okay well so far all my screenshots end up to be completly black xD. **I will send you a small file later when at home.**
And I will also try to move close all unnecessary windows and move everything away from the screen which might be in the way. Maybe that helps...

**Installation**
Well no you do not have to change anything on the Appimage. Just execute it and wait until Cura starts for the first time. After that you can close it and it should have created a folder in your Userfolder. 
~/.cura/ there is the ressource folder and everything else where you can copy the files as mentioned in your instructions.
Worked without any problems for me :)

Hope this helps for now. Will send you the promised file later. Thanks for investigating further.

timmehtimmeh, 2018-02-21:
Herzlichen Dank fur die Informatzion und die File - or is it das...my Deutch kann besser sein!

I've gone through the .g3drem file you provided and verified that indeed the data in the bitmap has only zeros for values which would indeed produce a black image.  I'm not sure why this is happening on your machine, but I can't seem to reproduce it using either an Ubuntu 17.10 system running from a thumbdrive, or on an ubuntu 16.04 system running on another computer in a virtual machine.  In both cases I get screenshots that have valid image data.  Nonetheless, I've made some improvements and increased the logging within the plugin and checked the changes into the develop branch.  Additionally, as you've done more measurements than I have, I've gone through and modified the density of dremel's filament within the marterial file.

If you want to check out the changes, feel free to overwrite the DremelGCodeWriter.py  located in 

`~/.local/share/cura/3.2/plugins/DremelGCodeWriter/DremelGCodeWriter`     with this version (you can right click the link to save the .py file):

https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/raw/develop/plugins/DremelGCodeWriter/DremelGCodeWriter.py  

I'm hoping at a minimum that with the new changes you get a generic Cura icon on the Dremel's screen.   With the increased logging I hopefully can figure out exactly where the zeros in the screenshot are coming from and maybe make a fix.

 After you get this plugin copied and run if you could post the last 30ish lines of cura's log located at ~/.local/share/cura/stderr.log - there should be some lines that look like this which is what I'm looking for:

```
2018-02-20 15:26:42,635 - ERROR - DremelGCodeWriter.DremelGCodeWriter.write [126]: Dremel GCode Writer - Black screenshot detected - using generic cura icon instead
2018-02-20 15:26:42,646 - ERROR - DremelGCodeWriter.DremelGCodeWriter.write [155]: Finished Writing Dremel Header.
2018-02-20 15:26:42,704 - ERROR - DremelGCodeWriter.DremelGCodeWriter._serialiseSettings [205]: No valid quality profile found, not writing settings to GCode!
```

Thank you again for helping me debug this - I sincerely appreciate your time!

SwapFaceL, 2018-02-22:
Hi again,

thanks for the reworked version.
So I replaced the python-Script and indeed it uses a cura icon now.

I also copied the (I think relevant) rows of the log:

```
[66]: 'info/send_slice_info' is turned off.
2018-02-22 10:07:19,018 - DEBUG - [(140039950141120)-MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [139]: Writing to Local File Desktop/test.g3drem in binary mode
2018-02-22 10:07:19,572 - ERROR - [(140039187052288)-Thread-3] DremelGCodeWriter.DremelGCodeWriter.write [121]: Dremel GCode Writer - Black screenshot detected - using generic cura icon instead
2018-02-22 10:07:19,577 - ERROR - [(140039187052288)-Thread-3] DremelGCodeWriter.DremelGCodeWriter.write [133]: Dremel GCode Writer - Black screenshot detected after scaling - using generic cura icon instead
2018-02-22 10:07:19,590 - ERROR - [(140039187052288)-Thread-3] DremelGCodeWriter.DremelGCodeWriter.write [162]: Finished Writing Dremel Header.
2018-02-22 10:07:19,726 - ERROR - [(140039187052288)-Thread-3] DremelGCodeWriter.DremelGCodeWriter._serialiseSettings [212]: No valid quality profile found, not writing settings to GCode!
2018-02-22 10:07:19,731 - DEBUG - [(140039187052288)-Thread-3] UM.FileHandler.WriteFileJob.run [61]: Writing file took 0.6948375701904297 seconds
2018-02-22 10:07:23,705 - DEBUG - [(140039950141120)-MainThread] UM.Qt.QtApplication.windowClosed [345]: Shutting down cura

```

metalman3797, 2018-02-22:
I can help with some of this issues you guys have. 

- For more accurate print times, ENABLE ACCELERATION CONTROL.  
[image]

timmehtimmeh, 2018-02-23:
Hi @metalman3797  -

Thanks for the info - I'll be the first to admit that I've spent almost zero time optimizing the Dremel printer settings in this repo.  If you've got a profile that you know works well for the Ideabuilder 3D20 and would like to export it and post it here I"d be happy to take a test it out and incorporate what I can into the default settings.

timmehtimmeh, 2018-02-23:
@SwapFaceL - thanks to your log it seems that the call to the QT library that's taking the screenshot isn't working on your machine (as opposed to the code that resizes the screenshot after it grabs it).  I was hoping that the original screenshot would be captured and that there was some error in the resizing section.  I'm not sure if I can do much about QT returning me nothing, and a quick google search didn't provide any useful information.  I'll take a look at implementing your suggestion sometime this weekend and post back when something is ready.

metalman3797, 2018-02-23:
I would be happy to update the profiles.  I have thousand of hours on our printers over here.  What other information are you missing on the *.g3drem format?  I can try to help fill in any of the blanks.

SwapFaceL, 2018-02-23:
@timmehtimmeh Thanks for the update. (Is the Update pushed in the Git already?) My printer is busy right now printing stuff for my car but I will definitively check out the preview image stuff. For updating the profile I am measuring right now the printing time of the current prints. Maybe we can adjust the profile then. I assume especially travel times, filling and support structures take more time than calculated. 
@metalman3797 Did you found the right settings already?

timmehtimmeh, 2018-02-23:
@SwapFaceL  - yes the update is committed to the develop branch here: https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/raw/develop/plugins/DremelGCodeWriter/DremelGCodeWriter.py  - feel free to test it whenever you like - as I said, I'll test it on mac and linux sometime this weekend.   Hopefully this fix will help you resolve this issue at which point I'll update the instructions and then make an official release and put the fix onto the main (master) branch.

@metalman3797 - Thanks for your offer to help!  Any and all settings that you know that work well for the IdeaBuilder would be welcome, from machine settings, to print profiles etc... The ideabuilder profile included in this repo only had the default temperatures modified (and as of yesterday I've enabled the acceleration control by default at your suggestion.  see: https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/commit/38d7add2306f85bad5a42763f44acac321c792ab )    Cura allows you to export alot of these settings, but screenshots would also work.  I've started a separate issue here to track the profile optimization: https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/issues/6 

_Re:_ which parts of the .g3drem file aren't yet understood - anything that says `"Magic #s"` in this section of the Readme: 
 https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin#technical-details-of-the-g3drem-file-format is a part of the .g3drem file format header that I don't currently understand.

metalman3797, 2018-02-23:
Yeah I already have the profiles done.  I am the lead software engineer at Dremel for the 3D team.  I create the profiles for our software as well as managing the firmware.  I have close to 10K hours on the 3D20's and probably another 20K hours on 3D40/3D45 since we started development so I am very familiar with this :)

timmehtimmeh, 2018-02-23:
@metalman3797  - it would awesome to have you help!   You definitely know more than me both about Cura plugin implementation and the g3drem file format (and python in general =)   Anything that you want to and are allowed to contribute would be most welcome

timmehtimmeh, 2018-02-24:
@SwapFaceL  - I've gone through and cleaned up the code that searches for an image in the same directory that you save the .g3drem file into.  Now there is no restriction on the capitalization of the filenames.  As promised I tested on Mac, Linux and Windows and on all three of my systems the implementation worked.  The latest code is here: https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/raw/develop/plugins/DremelGCodeWriter/DremelGCodeWriter.py  

If you get a chance to test in the next few days let me know how it works for you - if you don't get a chance, no worries!  I'll probably go ahead and make a version 0.3.0 release with the new features sometime next week

Cheers/Prost =)!

timmehtimmeh, 2018-09-30:
I'm going to close this issue now.  Feel free to reopen a new issue if anything isn't working, or you have any suggestions for improvement.  Thanks to both @SwapFaceL and @metalman3797  for your help - I sincerely appreciate it!

closed 2018-09-30 by timmehtimmeh