metalman3797/Cura-Dremel-Printer-Plugin #6: Improve Ideabuilder 3d20 profiles

timmehtimmeh, 2018-02-23:
The included Cura-Dremel-3D20-Plugin/resources/definitions/Dremel3D20.def.json has not been optimized .

SwapFaceL, 2018-02-23:
So today I printed an object with almost just infill. The predicted time was 5 hours but in reality it took about 5,5 hours.
I have another object (figure) printed with prediction 30 minutes but reality is 45 minutes.
Last object was estimated about 11 hours but took 1,5-2 hours more, but therefore had a lot of support structures.

So I guess what makes the prediction unreliable is a bit slower travel time, moderate slower infill and moderate slower support structures.

I will investigate further on this and maybe adjust the settings. If anyone observes something special about predicted time and real printing, please share.

metalman3797, 2018-02-23:
I dont have access to push any changes.  Tim can you grant me permission?

timmehtimmeh, 2018-02-23:
@metalman3797  - done... I did see your pull request come through...If possible please commit changes to the develop branch first - I'd like to keep the master branch as a stable branch and the develop branch as the development branch =)

metalman3797, 2018-02-26:
I was having a ton of problems pushing the changes through my works proxy.  It should be updated now.  We also need to create the low, medium and high quality profiles to associate with the 3D20

timmehtimmeh, 2018-03-01:
Thanks for updating the print profiles.  I took a swing at them from Simplify 3D's settings, but I'd bet that you know more about your printers than either I do or they do =)