metalman3797/Cura-Dremel-Printer-Plugin #20: Dremel 3D45 support?

pallthayer, 2018-04-11:
Hi. Thanks for the work you've done on this. I'm a Linux user and we just got a Dremel 3D45. I'm running the Ultimaker version of Cura. I was able to install this plugin without any issues but I'm wondering if this will properly prepare files for the 3D45. If not, is there any chance that that support will be built in some time?

timmehtimmeh, 2018-04-11:
Hi @pallthayer  - 

Unfortunately I don't have a 3D45 and am not too familiar with the differences between it and the 3D20 so I can't comment on whether or not this plugin will work for that machine.  There's nothing specific that is tied to a 3D20 regarding the writing of .g3drem files, other than it sets a few bits in the g3drem file indicating that the material is PLA, but I have no way of knowing whether or not the firmware in the 3D45 uses these bits or not.  The printer temperature is set both in the g3drem file header and in the gcode itself so i expect that you may be able to print other materials with no problem, but without access to a 3D45 I can't know for certain.

If the 3D45 does accept g3drem files then I suggest you give the 3D20 plugin a try as-is and report back (my first suggestion is to use PLA and the default 3D20 settings to start with).  

With a bit of work you should be able to modify the print bed sizes manually but if you have access to windows or mac then I suggest the following"

Dremel has their version of Cura that you can download [here](https://digilab.dremel.com/software).

Once you install the Dremel software on a windows or mac you can copy the 3D45 definition file from here (windows example below):
```C:\Program Files\Dremel DigiLab 3D Slicer\resources\definitions\Dremel3D45.def.json``` 
onto a thumb drive and paste it into 
```~/.local/share/cura/$CURA_VERSION/definitions/ ```

Then change the line in the original Dremel3D45.def.json that reads:
```"file_formats": "text/x-gcode",```
to read:
```"file_formats": "application/x-g3drem",```

save that out and you should be good to go (please note I haven't tested this and it may not work).  

Unfortunately without access to 3D45 I cannot commit to releasing a version of this plugin that supports the printer, as I wouldn't be able to test anything and bugfixing would be incredibly complicated and tedious.

Cheers
Tim

pallthayer, 2018-04-11:
Thanks for getting back so quickly. I ended up trying out the saved g3drem file using default settings for nylon (because that's what's currently in the printer). The machine displayed the file with a red circle and slash through it but started up anyway when I selected it to build. I then left the room for a while while it was heating up. When I got back, it displayed that the job was done but it hadn't done anything. I'll try out your tip with copying the def file from a Windows installation. Thanks.

timmehtimmeh, 2018-04-11:
Once you get the Dremel3D45 json file copied, you'll have to set up a new printer in Cura and select the Dremel3D45 option.  

The firmware inside the 3D45 may check the bits that tell which filament to use in the g3drem header against the installed filament and fail to print if the two don't match.  If you get a chance to test the output using PLA and it works (or doesn't) please let me know.

I'll take a look at the output of the Dremel Digilab software for the 3D45 and see if I notice anything markedly different in the file that it produces that would cause an error and will report back if I find anything significant.

janitz, 2019-05-30:
Thanks for the instructions!

also copy:
`C:\Program Files\Dremel DigiLab 3D Slicer\resources\meshes\Dremel_platform.stl`
to
`~/.local/share/cura/$CURA_VERSION/meshes/`

timmehtimmeh, 2020-06-09:
For anyone else reading this later, please see [this page](https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations) for where the CURA settings are stored for the various Operating Systems that Cura supports.  

For instance:

The `C:\Program Files\Dremel DigiLab 3D Slicer\resources\definitions\Dremel3D45.def.json` would be copied to:

**Windows**:  `%APPDATA%\cura\$CURA_VERSION\definitions\`
**Linux**:  `~/.local/share/cura/$CURA_VERSION/definitions/`
**MacOS**:`$User/Library/Application\ Support/Cura/$CURA_VERSION/definitions/`

and

> also copy:
`C:\Program Files\Dremel DigiLab 3D Slicer\resources\meshes\Dremel_platform.stl`
to
`~/.local/share/cura/$CURA_VERSION/meshes/`

On **Windows** the destination would be: `%APPDATA%\cura\$CURA_VERSION\meshes`
On **Linux**: `~/.local/share/cura/$CURA_VERSION/meshes/`
On **MacOS**: `$User/Library/Application\ Support/Cura/$CURA_VERSION/meshes/`

The same should be true for the 3D40 definition file.

timmehtimmeh, 2020-11-20:
Some additional info/files can be found [here](https://community.ultimaker.com/topic/25790-plugin-for-dremel-3d45/?do=findComment&comment=271263) and [here](https://community.ultimaker.com/topic/25790-plugin-for-dremel-3d45/?do=findComment&comment=271555].  I haven't tested these files, but someone might find them useful

timmehtimmeh, 2020-12-04:
One more update - another person [has made a repository](https://community.ultimaker.com/topic/25790-plugin-for-dremel-3d45/?do=findComment&comment=278130) with instructions on where to place the necessary files for the 3D45.  See [this github repository](https://github.com/Xcessity/CuraDremelSupport/)

timmehtimmeh, 2021-07-21:
@pallthayer  - I've recently updated (and renamed) the plugin to add in the 3D45 - feel free to give it a try.  I'd welcome any suggestions for improvements to the settings (particularly for the Dremel materials)

Cheers
Tim