metalman3797/Cura-Dremel-Printer-Plugin #26: Incompatible with Cura 3.4 curaplugin file no longer works

Swizzler121, 2018-06-27:
Just tried updating to Cura 3.4, was crashing when opened at first, then it said it had to wipe my settings, afterwards was unable to re-load curaplugin file, says unrecognized filetype. It looks like they've moved all plugins to some repository site they control.

timmehtimmeh, 2018-06-27:
Hi Swizzler121 - indeed - this is the same thing that I'm experiencing.  Every update of Cura so far has changed the plugin functionality a bit and things never quite work just after they release a new version of Cura.  In the case of Cura 3.4 it appears that Cura is moving in the direction of managing their plugins more tightly and the days of you installing the plugin from this repo may be limited.

Fortunately just this morning the Cura team has reached out to me to include this plugin in their repository (see https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/issues/25).  
I'll work with the them to get it included within their plugin browser, but that may take some time.  Until then I can only recommend that you stick with 3.2.1 or 3.3 if you need to print with the Dremel3D20.

timmehtimmeh, 2018-06-27:
Hi Swizzler121 

I dug some more into the crash, and pushed a change to the develop branch which temporarily fixes the crash.  If you still have your old configuration you can copy this Dremel3D20.def.json 
 file from here: https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/blob/develop/resources/definitions/Dremel3D20.def.json  over the old one.  On windows the old one is stored under C:\Users\<username>\AppData\Roaming\cura\3.4\definitions 

That *should* get you up and running again.

If you clicked "backup configuration" the old configuration will be backed up under C:\Users\<username>\AppData\Roaming\cura

Swizzler121, 2018-06-27:
Well I tried reinstalling 3.3.1, reloaded the cura config, then reinstalled 3.4, the .json file stopped it from crashing, but it popped up at the bottom saying the profile was corrupt and couldn't be loaded, asking me to reset to factory settings, and when I do that, it wipes the printer stuff. I'll just be patient and wait for the next plugin version.

timmehtimmeh, 2018-06-27:
Understood.  Thanks for your patience =)

Cura keeps a cache of printer definitions under here: C:\Users\%username#\AppData\Local\cura\3.4  it may be that the cache wasn't cleared out but I'm not sure.  

I can say that the following steps work for me (note I have Cura 3.3.1 with the 0.4.3 plugin installed and I don't uninstall Cura 3.3.1 when upgrading Cura, but that shouldn't matter):

1. Uninstall the crashing version of Cura 3.4
2. Delete the folder C:\Users\ %username% \AppData\Local\cura\3.4
3. Delete the folder C:\Users\ %username% \AppData\Roaming\cura\3.4\
4. Install the develop branch 0.4.4 plugin from here into Cura 3.3.1: https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/blob/develop/plugins/Dremel3D20.curaplugin
5. Install Cura 3.4
6. Launch Cura 3.4 (sidenote:  when you launch Cura the first time it copies over the plugin files from the previous versions which are by typically at: C:\Users\ %username% \AppData\Roaming\cura\3.3\ )

Note that the 0.4.4 plugin is the same as the 0.4.3 plugin with the exception that 0.4.4 replaces the printer definition json file with an updated one.  Everything else should be the same.

timmehtimmeh, 2018-07-04:
Hi @Swizzler121 

Thanks to the information provided me over [here](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/issues/25) I've been able to create a .curapackage file that you can drag-and-drop onto Cura version 3.4 and have subsequently released version 0.4.5 of the plugin: 
 https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/releases/latest

If you get a moment and would like to help me test this out feel free to give the latest release a go and report back if you have any issues.  Depending on the installation status of cura and the older plugin you may have to reset your configuration before installing this plugin.  If Cura crashes, or prompts you to reset to defaults it please click ok then drag-and-drop the .curapackage onto Cura afterwards. 

Please note that I'm still working on getting the optimized printer quality settings (the coarse/medium/fine settings on the slider,) working again after Cura's 3.3 update.  The settings shown below that are used with the 0.4.3 and 0.4.5 plugins are not optimized for the Dremel, and the optimized files that I do have are currently causing Cura to crash:
[image]

I'm hoping to get some information from the engineers at Ultimaker about these files, and re-enable the optimized settings with the next release.  Additionally I'm still working on getting this plugin integrated into the ultimaker online plugin browser - I'll close out #25 when that happens

Ghostkeeper, 2018-07-04:
> Cura keeps a cache of printer definitions under here: C:\Users%username#\AppData\Local\cura\3.4 it may be that the cache wasn't cleared out but I'm not sure.

Cura clears the cache upon every version upgrade. So when you upgrade to 3.4 the cache should be gone.

timmehtimmeh, 2018-07-07:
@Swizzler121  

I've fixed the problems mentioned above and now have made the 0.4.7 release which you can find [here](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/releases/latest).  I've also included some installation instructions for users who are upgrading from an older version of the plugin - please let me know if you have any problems.

Cheers!

Swizzler121, 2018-07-11:
Sorry! I was away from work on a trip, i'm back now, tested it, and everything works! thanks for being so speedy on this with a printer that doesn't get much attention otherwise, our library thanks you.

closed 2018-07-11 by Swizzler121

timmehtimmeh, 2018-07-13:
@Swizzler121  - Thanks for testing & reporting issues - I really appreciate you taking your time to help make open source software better.  I'm also truly grateful to be able to contribute my skills to help out a library - I hope that you and your patrons have nothing but success with their 3D prints, and that some of them are inspired to make great things!

FYI - as of this morning the plugin is now included in the Cura Toolbox - hopefully this will make setup for you even easier in the future =)

Cheers!

[image]