metalman3797/Cura-Dremel-Printer-Plugin #38: Support z-distance not correct on presets

Swizzler121, 2018-08-14:
You can close this if this a Cura issue and not an issue with this plugin but I haven't seen this mentioned on the Cura issues and it's caused enough issues I assume it would have been reported by now.

Since upgrading to 3.4.1 when using supports with default settings it will now print them far too close to the print and cause them to be fused to the model. You can fix this by tuning the support Z distance to be at the proper distance away but it appears the plugin by default sets them too close to the printed models.

timmehtimmeh, 2018-08-14:
Hi @Swizzler121 - Thanks for reporting this - I'll make a fix tonight.  Do you have a preferred z distance value that you've found works well?

Swizzler121, 2018-08-14:
I haven't found one that works as well as the old plugins, I've just increased what it's been auto-setting by .2mm and that seems to help, but still doesn't feel as "temporary" as the old settings. I've been thinking of starting an old instance of the software+plugin and see if the distances are the same or different.

timmehtimmeh, 2018-08-14:
@Swizzler121  Thanks for getting back so quickly.

This issue is probably related to the fact that some older versions of the plugin with older versions of Cura (cura ver 3.3  & 3.3.1) weren't using the presets that are included with this plugin, and were instead using the default Ultimaker values.  With the latest plugin I finally was able to fix that, and the plugin is back to using the values that are listed in my comment above.   The support_z_distance value was last updated in this plugin back in February, and was released with the 0.3.1 version of the plugin.

The "default" support_z_distance value that is used with the default Ultimaker values is 0.1mm across the board

```
at 0.06mm Layer height (Extra Fine Quality) the support_z_distance = 0.1mm
at 0.15mm Layer height (Low Quality) the support_z_distance  = 0.1 mm
at 0.2mm Layer height (Draft Quality, Fine Quality) the support_z_distance = 0.1 mm
at 0.4mm Layer height (Coarse Quality) the support_z_distance = 0.1mm
```

If you're willing and able to do some test prints and come up with a recommendation, or if you want to launch an older version of the cura/plugin combo and report back the values I'd be happy to implement the values as a default.

Swizzler121, 2018-08-14:
Okay, I tried printing it with all defaults at 0.2 layer height as shown in your screenshot above to make sure something weird I was doing wasn't messing things up and the supports fused at the bottom, and further up there was sagging in the supposedly supported material. I'm going to try it at 0.1 to see if that works, maybe it's fusing because it's not getting support and sagging into the support rather than being too close? will report back when this print finishes in ~30 mins.

Swizzler121, 2018-08-14:
Print finished, and that appears to have fixed it. changing the support z distance to 0.1 made the sagging over supported areas of the model go away and allowed a clean peel of the model from the supports. it appears making the gap too far causes the plastic to sag into the supports and fuse to them.

timmehtimmeh, 2018-08-14:
Thanks for testing - I sincerely appreciate your help =).  I'll issue a new version with the updated presets later tonight and work to get it included in Cura's toolbox (I've got a bunch of documentation changes that I want to release also).  I'll post again once the new version is released

timmehtimmeh, 2018-08-15:
@Swizzler121  - I've made Release 0.4.8 https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/releases/tag/0.4.8 that includes the support_z_distance = 0.1 for all quality profiles.  You can find the release [here](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/releases/latest).  

The first time I installed it the settings didn't update properly, which might be because Cura creates a cache of certain files, but when I tested a second time Cura did properly update - please feel free to test this out and please let me know if your version either does or does not update.  If it doesn't update I'll make some more changes to the plugin code where the files get installed to ensure that the cache gets cleared before the new files are copied in.

Cheers!

Swizzler121, 2018-08-31:
Hi, sorry for not writing back in so long. I was waiting for the repository to update to see if it just auto-updated my plugin. In the meantime though I noticed something else that might be a problem. Since the new cura version I've noticed these support interfaces printing. I'm not 100% sure if they always printed, but they seemed denser than they used to if they were there, then I discovered the support interfaces are defaulting to 100% density. I'm trying some test prints with them set at a lower density/thickness and also just off to see if I can find the setting that feels correct.

timmehtimmeh, 2018-08-31:
Thanks @Swizzler121 -  I apologize - I've been trying to get the plugin updated in Cura's plugin browser, but have so far not yet been successful.  The good news is that Cura is working on releasing version 3.5 sometime soon, and I will have to make another plugin release (ver 0.5.0 ) to support the new Cura.  If you find a combination of support settings that work for you, or any other settings that you'd like changed please feel free to post here and I'll incorporate them.  Sometime this weekend I'll try to take a look at what Cura's default settings are and let you know if I find anything significant.

Swizzler121, 2018-09-05:
okay, sounds good. I was about to say the same thing after reading their knowledgebase page on supports and doing several prints with different thickness interfaces.

timmehtimmeh, 2018-09-05:
Thank you again for your help!  

FYI - Last week I made the changes to disable the support interfaces and included those changes in the upcoming 0.5.0 release, which will only be compatible with the upcoming Cura 3.5 release.   I've submitted the 0.5.0 update to Ultimaker and they've tested it to validate that it doesn't crash.  I assume that when Cura 3.5 comes out the updated plugin will be included in their plugin browser, but I'll leave this issue open until I can validate.  

If you need these changes sooner feel free to reach out and I'll send you a compatible version that works with Cura 3.4.1

timmehtimmeh, 2018-10-03:
@Swizzler121 - Cura 3.5 is officially out of beta.  After installing you should be able to upgrade the Dremel3D20 plugin to version 0.5.2 (if it doesn't do so automatically), which includes the changes discussed in this bug.  Feel free to give it a try.

Swizzler121, 2018-10-05:
Okay, sounds good. I'll check them out. Thanks for all your hard work!

timmehtimmeh, 2018-10-06:
Thank you for your help as well!  As of this morning the plugin version 0.5.3 is now live in Cura's plugin browser.  I just ran a 16 hour print with tons of supports and didn't have a major problem removing the support material.  I'll leave this issue open for another week or two to give you a chance to test and I'll close it if I don't hear anything.

Feel free to keep making suggestions if there are other parameters that you find better settings for and want the defaults changed.

Cheers!
Tim

timmehtimmeh, 2018-10-31:
Closing.  

I believe that the issues reported here are fixed.  @Swizzler121 - feel free to reopen this if you feel that they're not.  Also please feel free continue reporting any other improvements that you think the plugin could/should have.  Thanks again for contributing to the project 😃!

closed 2018-10-31 by timmehtimmeh