metalman3797/Cura-Dremel-Printer-Plugin #44: Dremel 3D40 walls

TigerClaw33, 2018-11-06:
Hi, I am using the similar plugin for Dremel 3D40 with Cura3.5.1 but my print result is really bad. The walls thickness is like zero ... transparent

tim-schoenmackers, 2018-11-07:
Hi @TigerClaw33  
Thank you for taking your time to report an issue.  I'm sorry the plugin isn't working as expected for you =(  

Unfortunately I don't have a 3D40 and can't test any prints using it.  I can offer the following suggestions:

Depending on what you're printing, thin walls can be problematic - if the wall thickness of the file that you're trying to print is less than your layer thickness you may end up with zero thickness prints.  What do you see when you select the layer view in Cura?  If Cura doesn't show a wall there, then the gcode won't be generated and the 3d print won't happen.

Some things you can try:
1) Changing the wall thickness of the object you're printing - this may not be an option depending on what you're printing.
2) Changing the speed & layer height slider on the plugin - by default the plugin uses the medium quality settings, but there's a higher quality & slower setting that enables you to print thinner walls.
3) Changing to custom settings.  You can change the settings for yourself by selecting a "custom" setting option.  If you find settings that work better please feel free to reach out and share them better.  Note - the cura plugin "Cura Settings Guide" will be quite helpful if you go down this route.

From what I can tell, the Dremel3D40 supports standard gcode as well, so you should be able to export normal gcode using cura instead of exporting the .g3drem file.  Feel free to try that and see if it works better

Cheers!

TigerClaw33, 2018-11-07:
Hi, I changed all parameters but with Cura it's not working the walls are like transparent ... All lines are like that also the infill lines. I thing it's a hidden parameter or wrong configuration ... I cannot explain what it can be

TigerClaw33, 2018-11-08:
I copied all the files in the respective directory and chooses Dremel 3d40. the walls where normal on cura. but printing was catastrophical ... I settled all parameters like always ... but all went wrong ..

timmehtimmeh, 2018-11-08:
Hi @TigerClaw33 

Unfortunately without screenshots and/or the file that you're trying to print I'm afraid that I can't really help you more.  When you say that you settled all parameters like always, what exactly did you set.  What did you see when you select the "layer view" in cura?  Please post a screenshot of the layer view.

I'd encourage you to try downloading [the benchy print test](http://www.3dbenchy.com/download/) and import that into Cura to see if the walls are transparent in the plugin using benchy.

When you say you choose Dremel 3D40 how do you do that?  This plugin that I created is for the Dremel 3D**20**, not the Dremel 3D**40** and to my knowledge there is no way to select the Dremel 3D40 using the plugin.  The settings have also been optimized for the Dremel 3D20, and while they may work for the 3D40 I cannot test it, and there's no guarantee that the settings will work.  I can say that I've had many successful prints with the 3D20 using this plugin.

Please post screenshots to this page of the layer view, and give the benchy print test a try

timmehtimmeh, 2018-11-13:
@TigerClaw33 

I created an object with increasingly thin walls to test my theory that perhaps the walls of the object that you're trying to print are perhaps on the thin side.  The wall thickness in mm is noted below each wall:
[image]

I then imported this into Cura 3.5.1 using the latest release version 0.5.3 of the plugin.  

What I found surprised me a little bit, but may or may not explain what you're seeing with regards to the transparency.

The "Solid View" with the plugin and the Dremel 3D20 printer selected shows the object as expected:
[image]

however, if I switch to layer view, and select the "Line Type" as the color scheme, but do not hit the prepare button, then the object is transparent.  This makes sense, as Cura has not yet figured out the paths and layers.
[image]

If I do hit the prepare button, then I seethe following:
[image]

Note that the last three widths are transparent.  Basically somewhere between 1/2 mm and 1/4 mm thickness the settings on the plugin are now transparent. 

I then created a secondary object which varied from 0.5mm to 0.25mm by increments of 0.5 mm (wall thicknessses were 0.5mm, 0.45mm, 0.4mm, 0.35mm, 0.3mm, 0.25mm respectively)
[image]

What I found was that using the plugin and cura, wall thickness of 0.4mm or smaller do not generate any gcode and appear transparent

[image]

This makes sense - the nozzle of the Dremel 3D20 (and probably the 3D40 as well) is 0.4mm thick, and so the limit of what's printable should be right at 0.4mm thick.  Anything thinner would be thinner than the nozzle width, and as such will not print well.  I suspect that the object that you are trying to print has features which are less than or equal to 0.4mm thick.  As such they will appear transparent, and will not print well.  

If this is not the case, please either post screenshots of the layer view of the benchy object, or post a link to the object that you're trying to print so that I can take a look at it.

Cheers!

TigerClaw33, 2018-11-14:
[image]
This is my configuration. Normal Benchy normal settings

timmehtimmeh, 2018-11-14:
@TigerClaw33 - Thanks for posting the screenshot - the good news is that Benchy looks normal which leads me to believe that the plugin is behaving as expected - I don't see any transparent walls or other items that cause me any alarm for Benchy.   I suspect if you tried to print benchy on your 3D40 it would work

Because Benchy doesn't exhibit the transparent walls, I strongly suspect that there is a difference between the object that you tried to print that led to your initial bug report and the benchy object.  I suspect that the wall thickness of the object that led to your initial report is less than 0.4mm and as a result the walls are thinner than the nozzle diameter, and because of this Cura is not generating any layers for it.

TigerClaw33, 2018-11-14:
This is the object I wanted to print. Exactly this configuration. 

The Dremel Slicer is ok but I cannot print temptower with TweakZ Option. If I create t gcode it says "print done". Therefore I wanted to use the cura original software

timmehtimmeh, 2018-11-15:
I'm sorry - I think that perhaps an attachment didn't make it's way into your comment.  If you tried to attach a file, try zipping it up first and attaching it.

Searching for temptower I found this:  https://www.thingiverse.com/thing:2493504

which slices just fine in my version of Cura:
[image]

Regardless -  As I mentioned in a prior comment the Dremel3D40 supports standard gcode - you can set up a custom printer and not use this plugin, and simply use the cura slicer to export standard gcode.  Feel free to try that to see if it suits your needs.

timmehtimmeh, 2018-11-30:
@TigerClaw33 

Did you ever figure out the issue.  

Using the temptower linked to above
[image]

and changing the temperatures 5 degrees C every ~34 layers  (first layer transitions at layer 42) 
[image]

as indicated by this video:
https://www.youtube.com/watch?v=Wkegh9WRaxs

I was able to slice and save the file with no problem and no transparency.  This was done using the Dremel Plugin version 0.5.3 set at the "medium" quality setting (layer height 0.2) with Cura version 3.6 

Looking at the gcode section of the .g3drem file that was produced for this test in a text editor, I can find the places where the gcode temperature transitions happen.  The gcode commands for extruder temperature are [M104 and M109](https://www.simplify3d.com/support/articles/3d-printing-gcode-tutorial/#M104-M109).  By searching for the M104 command, I can find places where Cura & the Change At Z post processing script modify the temperature.

For instance, here is layer 1, which sets the temperature to 220C
[image]

And here is the next transition to 215C at layer 42
[image]

Here is the next transition to 210C at layer 76
[image]

This goes on in a similar fashion until it hits the final bridge of the temperature tower at layer 278 where it sets the top bridge's temperature to 180C
[image]

One other way that Cura can make the layers appear transparent is if you're in layer view, and you move the layer slicer down.  This would cause any layer above the set layer to appear transparent.

[image]

I've attached the sliced .g3drem temperature test file that I've made below, and will close out this issue as I'm unfortunately unable to reproduce your problem.  I hope that the above information & following file is helpful to you.

[D_ksr_fdmtest_v4.zip](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/files/2632306/D_ksr_fdmtest_v4.zip)

If you're able to provide me any more info that helps me to figure out what's going on with your system, please feel free to re-open the issue, but unfortunately right now I don't have the info necessary to

closed 2018-11-30 by timmehtimmeh