metalman3797/Cura-Dremel-Printer-Plugin #47: Blurry texts in Preference dialog

diegopradogesto, 2019-01-08:
When I open the preference dialog, I see some blurry texts. I use a secondary screen.

[image]

In Cura we used to have issues like that and we fix them by rounding the width and height (using Math.round) of some elements in qml. For instance if dividing by 2 you can get half a pixel and so the blurry text appear. You can also try to fix it by using `renderType: Text.NativeRendering` in all the labels in the qml files.

timmehtimmeh, 2019-01-09:
@diegopradogesto - Thank you very much for reporting this & for the tips on how to fix - I sincerely appreciate it =).

I've committed some changes to the qml on the develop branch [here](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/commit/8de069656a0da2d32725cf95a41982e87c255a35) that will hopefully fix the issue that you're reporting.   Unfortunately I haven't yet been able to duplicate the issue myself.  Changing the display scaling on windows seems not to show this behavior at 100%, 125%, 150% and 175%.

I'll keep working on it to see if I can find a system to reproduce the text scaling issue on - once I can confirm that it's fixed I'll submit an update to the plugin.

Cheers!

diegopradogesto, 2019-01-09:
I'll try to reproduce it again today in my computer. I'll let you know if the issue is still there.

diegopradogesto, 2019-01-11:
Apparently it didn't work for all the text in the checkboxes. I made some comments in the commit and removing the `qsTr` in the checkboxes' text seems to work for me. Also adding a `renderType` property does nothing in the checkbox, it is only available in Labels.

timmehtimmeh, 2019-01-12:
Thanks for taking the time to help me test this @diegopradogesto 

Per your testing I've removed all ```qsTr``` calls within the qml file, and cleaned up the preferences dialog (commit is [here](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/commit/43412a920cd3a48e9a36764e63313c2b48a497fa)).  I'll test on a few more machines this weekend, both with and without external monitors and hopefully I'll be able to find one that I can duplicate the behavior on to verify that I fixed it.

The new preference dialog should look like this if everything renders correctly:
[image]

Cheers!

diegopradogesto, 2019-01-22:
Looks fine now. I'm closing this issue then.

BTW, this is an off-topic but you probably realised that your plugin is not available in Ultimaker Cura 3.6 anymore. That is because we had and internal issue that didn't allow to use the same plugin for different SDK versions. However we did a fix for this and maybe you can resubmit your plugin with those changes. You can take a look in this documentation:

- https://github.com/Ultimaker/Cura/wiki/CuraAPI-and-SDK-Versions
- https://github.com/Ultimaker/Cura/wiki/plugin.json

Please let me know if I can help with that.

closed 2019-01-22 by diegopradogesto

timmehtimmeh, 2019-01-23:
@diegopradogesto  - I cannot thank you enough for reporting & subsequently helping to test this issue out.  - I sincerely appreciate your time & efforts.  I've credited you as a contributor in the README - Thank you!

I ([or rather, one of the users of this plugin](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/issues/48)) had noticed that the plugin disappeared from the Cura 3.6 marketplace and I figured that I had just released too soon, so thanks for the information.  Typically I try to follow the style of the [GcodeWriter plugin](https://github.com/Ultimaker/Cura/blob/4.0/plugins/GCodeWriter/plugin.json) but that plugin doesn't seem to use the new ```supported_sdk_versions``` field, and I'm having difficult getting Cura 3.6 to recognize my new plugin.  Here's what I've tried so far:

1) Adding the the line ``` "supported_sdk_versions": ["5", "6.0.0"],``` to my plugin.json while leaving the ```"api": "6.0",``` line alone - Cura 3.6 didn't recognize the plugin as installed
2) Adding the the line ``` "supported_sdk_versions": ["5", "6.0.0"],``` to my plugin.json while leaving the ```"api": "5",``` line alone - Cura 3.6 didn't recognize the plugin as installed
3) Adding the the line ``` "supported_sdk_versions": ["5.0.0", "6.0.0"],``` to my plugin.json while leaving the ```"api": "5",``` line alone - Cura 3.6 didn't recognize the plugin as installed

Internal to the 0.5.6 plugin I've added lines that call Cura  built in Snapshot() functionality (which I believe is new to cura 4, but feel free to correct me) and removed my old hacked screenshot code, so I'm probably best off leaving the ``` "supported_sdk_versions": ["6.0.0"],``` as version 6 only.  In a month or so once Cura 4 comes out this shouldn't be an issue, and I doubt that there are too many inconvenienced users -  but I'd be willing to re-release the plugin for 3.6 if I can test it before I release - if you've got any tips I'd be happy to hear them 

Thanks again for your help

diegopradogesto, 2019-01-23:
Thanks for the credits, I think adding more functionality to Cura is always better for us and also for you. So, thank you for the contribution.

Asking your questions:

- Yes, you're right, the GCodeWriter plugin is not really updated and still uses the deprecated "api" field. We have to change it.
- To make your plugin work with both 3.6 and 4.0, you need to do this in your plugin.json (that is something like option 3, maybe you need to keep the "api" using a integer):
```
{
   ...
   "api": 5,
   "supported_sdk_versions": ["5.0.0", "6.0.0"]
   ...
}
```
- The Snapshot functionality was added before 3.5, so it should work in your plugin. If not, you can do two things:
1. Keep your plugin only available for 4.0, using only `"supported_sdk_versions": ["6.0.0"]`.
2. In your code, you can check for the SDK version and use an if-statement to do different things depending on the version. 

Let me know if that works and if I can help you further.

diegopradogesto, 2019-01-24:
You only need to do one release with 
```
{
    ...
    "api": 5,
    "supported_sdk_versions": ["5.0.0", "6.0.0"],
    ...
} 
```

The problem in Cura 4.0 beta is that the feature was not ready, that's why it doesn't work. I can provide a nightly build for you or just upload it in the contribution portal and I will test it in both versions.

timmehtimmeh, 2019-01-24:
Thanks again for all your help - I've submitted the plugin - feel free to let me know if you find any issues.  Cheers!