metalman3797/Cura-Dremel-Printer-Plugin #48: Cannot install plugin manually or via Marketplace

HartAtttack, 2019-01-10:
Hello,

I'm on a 3D20 printer, using Windows. 

I installed your 3d20 plugin only a few days ago from the marketplace, but after a reset because of an error I went to re-install and noticed it's not there anymore. 

After that, I tried a manual install of 0.5.3 on Cura 3.6.0 and I'm getting an error every time. 

"Dremel 3D20 plugin experienced an error installing the files."

I can't install this plugin no matter what I do! The one time I used your plugin it was great! Thanks,

timmehtimmeh, 2019-01-11:
Hi @HartAtttack 

I just published the 0.5.5 plugin for Cura 4.0, which unfortunately took down the 0.5.3 plugin from the marketplace.  

I just installed Cura 3.6 on this machine, and downloaded the [Cura-Dremel-3D20-0.5.3.curapackage](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/releases/download/0.5.3/Cura-Dremel-3D20-0.5.3.curapackage) from the releases page (linked).  Upon dragging the .curapackage file over I received a restart message:

[image]

and then when I went to restart the Dremel was available
[image]

To get the plugin working there are a few options
1)  Download and install the Cura 4 beta, then install from the marketplace - this should work
2)  Manually clear out the old plugin - to do so (assuming you're on windows) open 
```c:\Users\<USERNAME>\AppData\Roaming\cura\3.6\plugins``` and delete the Dremel3D20 folder from there
You might also need to delete the following files
```C:\Users\<USERNAME>\AppData\Roaming\cura\3.6\definitions\Dremel3D20.def.json```
```C:\Users\<USERNAME>\AppData\Roaming\cura\3.6\extruders\dremel_3d20_extruder_0.def.json```
```C:\Users\<USERNAME>\AppData\Roaming\cura\3.6\materials\dremel_pla.xml.fdm_material```
```C:\Users\<USERNAME>\AppData\Roaming\cura\3.6\meshes\dremel_3D20_platform.stl```
as well as the folder located at
```C:\Users\<USERNAME>\AppData\Roaming\cura\3.6\quality\dremel_3d20```

That should reset your cura back to the default without any of the dremel plugin within it.   From there you should be able to reinstall the plugin without any issues.  If that still gives you problems, you can rename the entire ```C:\Users\<USERNAME>\AppData\Roaming\cura\3.6``` folder to ```C:\Users\<USERNAME>\AppData\Roaming\cura\backup3.6``` and then launch Cura again.  Cura will re-create this folder with defaults and you should be able to re-install the plugin from there

If you could also post your cura.log file located at
```C:\Users\<USERNAME>\AppData\Roaming\cura\3.6\cura.log``` I can take a look at it and see if I can figure out which file is causing issues.

Cheers

HartAtttack, 2019-01-16:
Thanks for your prompt reply, much appreciated.

I ended up switching to the 4.0 beta with the new version of your plugin and I have been successfully using it. Thank you. 

For the record, I had no idea you could just drag and drop the package file into the software. I was trying to copy the whole folder into my program files... not sure if that makes a difference.

closed 2019-01-16 by HartAtttack