metalman3797/Cura-Dremel-Printer-Plugin #54: Crashes when trying to export g3drem file

brendanwilding, 2019-06-01:
Crashes when trying to export g3drem file. No messages given. Able to export gcode fine.
Can't work out where log files are on Linux AppImage.

timmehtimmeh, 2019-06-06:
@brendanwilding  - Sorry it took me so long to respond.  For some reason I did't get an email for this one.

I assume you're on linux, correct?   What version of Cura and what version of the plugin are you using?

timmehtimmeh, 2020-02-24:
Closing due to lack of activity, because I can't duplicate, and the code has been updated.  @brendanwilding - feel free to reopen if the issue still persists with Cura 4.4 and plugin version 0.6.1

closed 2020-02-24 by timmehtimmeh

timmehtimmeh, 2020-04-05:
@brendanwilding  - I just released a fix for a crash in certain circumstances if the screenshot failed - please try with version 0.6.2 and if you still experience a crash feel free to report back