metalman3797/Cura-Dremel-Printer-Plugin #55: Won`t pick up connection to windows pc

jimbo2459, 2019-06-06:


timmehtimmeh, 2019-06-06:
Hi @jimbo2459 

I assume you're trying to connect the 3D20 directly to cura via a USB cable - correct?  

This plugin does not and will not support a direct connection via USB to the Dremel3D20.  It only supports exporting .g3drem files, which you can then place on a SD card that you can put into your Dremel 3D20.  Please see the [Using the Plugin](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin#Using_the_Pluginf ) section of the README for a detailed description of how to use the plugin.

If you need to support a direct connection via USB cable please give the [Dremel Digilab](https://digilab.dremel.com/3D-software) a try - the official Dremel software is based on Cura as well - and they (probably, I haven't checked) support direct USB connections

jimbo2459, 2019-06-06:
Hi Tim thanks for the reply you are correct I was trying to connect the printer direct to cura I was hoping there would be a workaround to get cura to recognize the 3d printer.

timmehtimmeh, 2019-06-06:
Hi @jimbo2459 - I'm sorry, but unfortunately not =(   I have no information about the Dremel3D20's USB connection and what data/format the printer expects over the USB connection, so unfortunately I can't support it.

closed 2019-06-06 by timmehtimmeh