metalman3797/Cura-Dremel-Printer-Plugin #60: Cannot find plugin in Cura Marketplace & no manual install instructions

semiconducted, 2019-12-03:
Cannot find plugin in Cura Marketplace.

I tried to install this manually by moving the files into the correct folders, but no .dgrem file type appears in 'save file' dialog and manually changing file type results in Dremel 3D20 crashing when exported file is loaded.

MacOS: 10.15.1 (19B88)
Cura: 4.2.1

Is this an issue local to just me?

[image]
[image]

timmehtimmeh, 2019-12-03:
@semiconducted - Unfortunately Cura doesn't have a stable API, and each time they make changes older versions of cura lose access to the plugin in the marketplace.  Please download the appropriate version of the .curaplugin from the releases page here:  https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/releases

Then drag and drop the .curaplugin onto your running cura instance

semiconducted, 2019-12-03:
Thanks for the incredibly quick reply!
It turns out it WAS a local Cura issue just for folks like me running an older version. 
I had reinstalled Cura before, but the key this time was manually deleting all references to it, i.e. everything in the Application Support folder and reinstalling from scratch. 

I'm now on Cura v 4.4.0 and install worked just as you wrote in the README.md.

Thanks!
[image]

closed 2019-12-03 by semiconducted

timmehtimmeh, 2019-12-03:
Glad to hear you resolved the issue & thanks for the update - it will help anyone else having the issue.  Cheers!