metalman3797/Cura-Dremel-Printer-Plugin #61: Dremel D345

Walter-58, 2020-02-23:
Hello,
can the plugin also be used for the Dremel D340 or D345?
How big would the adjustments be?
I saw it is Java Script.
I can use C ++ and C #.
greeting
Walter

tim-schoenmackers, 2020-02-23:
Hi @Walter-58 

> can the plugin also be used for the Dremel D340 or D345?
> How big would the adjustments be?

Good question - see https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/issues/20 for a discussion on how to get the 3D40 or 3D45 working.  If you need more support than what's discussed there you might be able to reach out to those users and see if they can help you.  Please note that I don't have a 3D40 or 3D45, and as such won't include the capability in this plugin (supporting a printer that I don't own is not something that I can commit to)

> I saw it is Java Script.
> I can use C ++ and C #.

The plugin is written in python, as that's what Cura uses for plugins.   In practice you could modify the plugin to call into c++ dlls, and write a python wrapper around the dll, but at some point you have to write a little python if you want to interface with cura.

If you're only interested in writing some C++/C# to make a .g3drem file then you should be able to do that - please see the [Technical Details of the .g3drem File Format](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin#Technical_Details) section of the README

timmehtimmeh, 2020-02-24:
Also @Walter-58  - if you'd like to make a Cura plugin for the 3D40 or 3D45 please feel free to fork this plugin (it's open source under GPL 3.0) and use it as a base to start with - I don't think there are so many changes required to get the basic plugin working (where you write to a sdcard...network connections will provide a secondary set of challenges/issues).  You're certainly not the only one who wants a Cura plugin for the 3D45, and if you have the skills to write code I'm sure that people would appreciate having a 3D45 plugin in the Cura Plugin store.  

If you do make a plugin please credit the Cura gcode writer and this plugin, but by all means please feel free to write code based on this repository.

closed 2020-04-03 by timmehtimmeh

timmehtimmeh, 2020-12-04:
I just saw that [another person has made a repository](https://community.ultimaker.com/topic/25790-plugin-for-dremel-3d45/?do=findComment&comment=278130) to enable the 3D45 - See [this link](https://github.com/Xcessity/CuraDremelSupport/) for the instructions

Walter-58, 2020-12-04:
Danke für die Info. 
Werde ich ausprobieren. Hatte mir mit einem D320 und angepassten Abmessungen weitergeholfen. Funktioniert auch.
Grüße
Walter

Von: Tim Schoenmackers
Gesendet: Freitag, 4. Dezember 2020 18:12
An: timmehtimmeh/Cura-Dremel-3D20-Plugin
Cc: Walter-58; Mention
Betreff: Re: [timmehtimmeh/Cura-Dremel-3D20-Plugin] Dremel D345 (#61)

I just saw that another person has made a repository to enable the 3D45 - See this link for the instructions

timmehtimmeh, 2021-07-21:
@Walter-58  - over a year later I've released a renamed version of the plugin that includes 3D40 and 3D45 support - as always I welcome suggestions for improvements.

Schoene Grusse!

Walter-58, 2021-07-21:
👍

Super vielen Dank

Gruß

Walter Haas