metalman3797/Cura-Dremel-Printer-Plugin #63: Cura crashes: 'NoneType' object has no attribute 'save'

Ghostkeeper, 2020-03-26:
We got this issue report through Cura's issue reporter. Cura was crashing somewhere in this plug-in. I thought I'd share it with you guys.

AttributeError Dremel3D20.Dremel3D20 in getBitmapBytes
====
'NoneType' object has no attribute 'save'

```
AttributeError: 'NoneType' object has no attribute 'save'
  File "X:\4.5-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 1023, in event
  File "X:\4.5-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 456, in event
  File "X:\4.5-exe\build\inst\lib\python3.5\site-packages\UM\Event.py", line 189, in call
  File "X:\4.5-exe\build\inst\lib\python3.5\site-packages\cura\Utils\Threading.py", line 34, in _handle_call
  File "Dremel3D20\Dremel3D20.py", line 473, in getBitmapBytes
    if not self._snapshot.save(bmpData, "BMP"):
```

It seems that `self._snapshot` can contain `None`?

timmehtimmeh, 2020-03-26:
Thanks for the report @Ghostkeeper - I haven't been able to duplicate this on my side but I've added additional checks for None in the plugin and submitted an update via the Ultimaker portal.

Cheers!

timmehtimmeh, 2020-03-26:
I've submitted version 0.6.2 via the Ultimaker portal that contains the fixes.

Cheers!

timmehtimmeh, 2020-04-02:
The new plugin is now published in the Ultimaker Marketplace - thanks for the helpful bug report @Ghostkeeper!

closed 2020-04-02 by timmehtimmeh