metalman3797/Cura-Dremel-Printer-Plugin #69: I have the old software back from 2015, which allows you to print over USB cord.

CodedBagol, 2020-12-31:
I can upload it if you want. Don't know why they would remove this feature except for planned obsolescence.

timmehtimmeh, 2020-12-31:
Hi @CodedBagol  -  

It's awesome to have users like you who want to help and make the plugin better - thank you - I really appreciate it!  

I wasn't even aware that Dremel had removed the USB connection capability, but now that you mention it I don't see any discussion on their [how to use digilab](https://digilab.dremel.com/digilab-3d-slicer-basics) page.  

I have a copy of the `Dremel 3D_0.9.20150806_win.exe` installer that I downloaded in 2017 that supports USB connection to the printer - opening it for the first time in years is a fun blast from the past 😃 :

[image]

Regarding As I say in the [Frequently Asked Questions](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin#FAQ) section the plugin doesn't currently support the USB connection:

> **Question**: Does this plugin support USB-connected printing/Do I need an SD card 
>**Answer**: The plugin only supports creating .g3drem files and using an SD card to get the files into the printer. It does not support direct USB communication with the printer. Maybe if the author gets really bored one day, but it's unlikely

The main issue with, and the reason I haven't implemented the USB connection is that I'd have to reverse engineer the serial communication protocol, including the error conditions that may not happen very frequently.  I'm also not sure if the driver that is installed with the Dremel3D software is required, but if it is then I'd have to implement a driver as well.  It's not out of the realm of possibilities, but currently (for me) the amount of work it would take to implement this outweighs the benefits that it would bring.  That said - if you have a compelling use case for using the USB connection I'd love to hear it.

closed 2021-02-15 by timmehtimmeh

timmehtimmeh, 2023-08-04:
@Oscar-T24 - I might still have a copy but I'll have to look for it.  If I find it I'll let you know.  Perhaps @metalman3797 has a copy.

timmehtimmeh, 2023-08-04:
@Oscar-T24 - I need to pull out a bunch of old USB sticks to look for it...won't be able to do get to this until early next week.  There are instructions on how to set the 3D20 up with octoprint,  which uses a USB cord to send the gcode to the machine.  If you're interested in that see the instructions here: https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/tree/stable/docs/octoprint

Oscar-T24, 2023-08-04:
Thanks !! I was looking for a way to directly connect to the 3D20 through USB, so octoprint would be more suitable right ? ( I already use your 3D20 plugin on cura but I have to constantly plug and unplug my sd card, not very convenient...) All I need so far to set up octopart is a Pi ? Thanks!!

timmehtimmeh, 2023-08-04:
@Oscar-T24  -  I'd definitely recommend giving octoprint a try before using the old software.  The old software did not have the ability to create any supports which severely limited what you could print with it.  FWIW - the lack of supports is why I initially wrote this plugin.

I haven't tried octoprint in a while, and unfortunately I don't have a 3D20 anymore, so I can't really test it, but when I wrote those instructions two years ago it did work...feel free to give it a try.  You'll need a pi (and possibly a wifi dongle if the pi you're using doesn't have wifi built in)...the instructions have the requirements for what you'll need.

timmehtimmeh, 2023-08-09:
@Oscar-T24 - I found a copy of the `Dremel 3D_0.9.20150806_win.exe` executable.  You can grab it [here](https://file.io/GMHHz57o9Yrg).  **Note** The link is only valid for a week.

Oscar-T24, 2023-08-09:
Thank you so much! However the executable doesn't seem available on your file.io because the page says the file has been deleted :/

timmehtimmeh, 2023-08-09:
Try [this](https://file.io/R7tCf2svMvLN) and let me know if it works

Oscar-T24, 2023-08-14:
Thank you ! 
I could successfully print a 3d model from my computer, through the USB cord. I will try octoprint as well

duanecr, 2024-07-22:
> Try [this](https://file.io/R7tCf2svMvLN) and let me know if it works

Would it be possible to get a new link to download this file?  I'm trying to connect via USB and cannot get the driver.

timmehtimmeh, 2024-07-22:
@duanecr - [here](https://file.io/h5Yr7yFP5Dkf).  Cheeers

duanecr, 2024-07-22:
Thanks!

You are receiving this email because you were mentioned.

jglozano, 2025-01-20:
@timmehtimmeh apologies for asking but can you share again? I'm stuck in the same situation as others. I appreciate your response.

timmehtimmeh, 2025-01-20:
[here you go](https://limewire.com/d/2edb357c-7e34-445a-9845-d6c6644d47e7#yXXc0NYnJhrKHWT9JDWTpm23c7bweUWLSXta1v4R_tI)

jglozano, 2025-01-20:
@timmehtimmeh thank you very much! My 14yr old has been wanting to print a few things and I've been struggling for weeks trying to find the right software to use.

timmehtimmeh, 2025-01-20:
@jglozano - the software I uploaded is rather old and doesn't allow basic features like adding supports to your prints.  I'd only recommend it if you _really_ need the USB direct connection.  If you're just looking for a solution to slice files for the Dremel and you have a spare thumbdrive I'd recommend giving Cura with this plugin a try.  To use it simply install Cura and go to the marketplace and install the Dremel plugin (both Cura and the plugin are completely free)

jglozano, 2025-01-20:
thanks @timmehtimmeh! I was planning on doing the update of the firmware since the device is the same as when I bought it.

SuperJames888, 2025-02-10:
@timmehtimmeh Can I get the firmware as well?

timmehtimmeh, 2025-02-11:
@SuperJames888  - firmware for the 3D20 is [here](https://3pitech.com/pages/3d20-support-firmware-documents)

SuperJames888, 2025-02-11:
@timmehtimmeh So sorry, I meant the Dremel 3D_0.9.20150806_win.exe

timmehtimmeh, 2025-02-11:
@SuperJames888 [here](https://limewire.com/d/cfca0558-a49e-4cff-9493-6b580525de7b#TbvRM0NP0Jov_hUc73PptgzrRNrevbWUwfLODqB4K8s)

timmehtimmeh, 2025-03-06:
[here](https://limewire.com/d/7e5a14d1-37fd-4717-99ca-13aa173a61bc#Z_qOTlZ9s8-yV3AGIJ7nSqnr1vz0TdkjMDQcX_1qO80)

Larry192003, 2025-03-14:
@timmehtimmeh could I possibly get the firmware Dremel 3D_0.9.20150806_win.exe as well? I cannot for the life of me find the driver.

timmehtimmeh, 2025-03-14:
[here](https://limewire.com/d/e5a5aedf-193e-4850-bdb1-a973fa37d392#lpDnOXYztnDHrhqZ73ffjJbkRidbVFJhHUmTmrzlVDg)

Larry192003, 2025-03-14:
Thank you so much!!

corbyborb, 2025-03-22:
could you send me the Dremel 3D_0.9.20150806_win.exe as well? Been trying everything just to get the printer to show up again on the PC. showed up and was able to connect for the dremmel firmware 20210804 but hasn't registered since. Thank you

SuperJames888, 2025-03-23:
@corbyborb Just so you know, it seems the latest Dremel 3D20 firmware disables the USB. So make sure you do not update the firmware.

I am planning on making a permanent link, since more and more people want the firmware now.

ADSG95, 2025-05-12:
@timmehtimmeh I'm so sorry, but do you still have a copy of the installer?, I'm trying to update to the latest firmware my 3D20, but W11 don't recognize my printer :(

timmehtimmeh, 2025-05-12:
@ADSG95 - [here](https://limewire.com/d/epyLw#WQcH3u3DZz)

ADSG95, 2025-05-12:
@timmehtimmeh Thank you man, really appreciated; faster than Dremel support!

timmehtimmeh, 2025-07-20:
@vikjg - [here](https://limewire.com/d/8ykYH#HPZl23Cdrd)

vikjg, 2025-07-20:
@timmehtimmeh Thank you!

ckhiga, 2025-10-07:
@timmehtimmeh could you share the file again? kind of feels embarassing...

timmehtimmeh, 2025-10-07:
@ckhiga  - [here](https://limewire.com/d/cHAId#NTKUMXMDyX)

hughkim74-droid, 2025-10-26:
@timmehtimmeh Could you share the file one more time? it is really hard to find on web. Thank you alot.

SuperJames888, 2025-10-26:
The file is uploaded to the Internet Archive, due to me seeing the amount of people who are requesting it. Here is the link for the file. [https://archive.org/details/dremel-3-d-0.9.20150806-win](url)