metalman3797/Cura-Dremel-Printer-Plugin #70: Unable to select

CeeCeePea, 2021-01-22:
I am not sure if the issue has to do with the plugin but after a while I cannot select the object on the screen. I have to restart the program and reload the file to be able to select the design.

timmehtimmeh, 2021-01-22:
Hi @CeeCeePea - most likely this is a Cura issue, and not an issue with this plugin specifically.  It appears that others have had similar issues with Cura- see [here](https://community.ultimaker.com/topic/27891-problem-selecting-items-in-cura-40/) and [here](https://community.ultimaker.com/topic/33478-cant-select-any-objects/) including users who report the issue even with no plugins installed.  Some users have reported the issue up to and including Cura ver 4.8

The Dremel plugin code is only be called when Cura is writing a .g3drem file out - other than that all user interaction is handled by Cura.

CeeCeePea, 2021-01-22:
Thank you for your prompt response and help. 

CeeCee

nallath, 2021-01-26:
I'm fairly sure that this is a Cura issue since it also happens with other plugins.

closed 2021-01-27 by timmehtimmeh