metalman3797/Cura-Dremel-Printer-Plugin #71: A version for 3D45 would be great

Rally037, 2021-01-22:
Very nice app, would love to have this for the 3D45 too.

timmehtimmeh, 2021-01-22:
Hi @Rally037 

Unfortunately I don't own a 3D45 and therefore I can't commit to supporting the printer.

Here's a snip from the README on this plugin's page:
**Question**: Does/Can this plugin support the Dremel 3D40/Dremel 3D45? Answer: For the 3D45 please see [this github repository.](https://github.com/Xcessity/CuraDremelSupport/) Specifically for this plugin and the 3D45, wth some work maybe you can get it working- see ["Issue #20"](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/issues/20) for a more complete answer. The author doesn't own and doesn't have access to a 3D40/3D45, and cannot commit to supporting any machine that they don't have direct access to. Some users have been successful at getting the machine working with a modified version of this plugin, however it will require some work.

Rally037, 2021-01-22:
Hi there,

Thanks for the note.

If I could get you one, would you?

Thanks!

JP

timmehtimmeh, 2021-01-22:
@Rally037 - 

Wow - that's very generous of you!

I'd certainly be happy to give it a go, however I'd like to make sure that your expectations are properly set beforehand.  The 3D45 is a much more complex machine than the 3D20, and currently not all of the features of the 3D20 are supported (one example: you can't send a print to the 3D20 over USB using this plugin - you have to save a .g3drem file to an SD card).  There are a few reasons for this that I outline in the README (below) and in some of the issues, i.e. [here](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/issues/69) - most of it boils down to the fact that it works well enough for my use case that spending the extra (enormous amount of) time to reverse engineer Dremel's implementation isn't worth it (to me).

> **Question**: Does this plugin support USB-connected printing/Do I need an SD card **Answer**: The plugin only supports creating .g3drem files and using an SD card to get the files into the printer. It does not support direct USB communication with the printer. Maybe if the author gets really bored one day, but it's unlikely

Similar to how the 3D20 plugin works, if I were writing a 3D45 plugin, the plugin would only support the following:
1. Writing a .g3drem/.gcode file to a USB stick for printing.  
2. Automatically installing the printer definiton files to the correct directory in Cura
3. Automatically installing the extruder files to the correct directory in Cura
3. Automatically Installing the build platform stl  to the correct directory in Cura
4. Automatically installing the necessary files, including the materials (PLA, ECO-ABS, Nylon, PETG)  to the correct directory in Cura
5. Automatically installing the quality settings (0.05, 0.1, 0.2, 0.3 and 0.34 layer heights) to the correct directory in Cura
6. The plugin would be initially installable via drag & drop followed by the Cura marketplace pending Ultimaker Marketplace testing & approval.

The 3D45 plugin would **not** support support the following:
1. printing via USB
2. printing via the Dremel Cloud
3. printing via the network connection
4. Any other "fancy" feature not listed above.

There may be something I could do about opening an interface to the camera, or possibly implementing network printing but without poking at I would say that by default it's not supported.

Additionally (just to be clear) I'd need to keep the printer permanently to ensure that future releases of Cura and the plugin were compatible.

With all of that said - you can do most of the items that I'd support manually right now using the [files in this repo](https://github.com/Xcessity/CuraDremelSupport/).

if that sounds good then let me know and we can work out the details...if not - no worries!

Cheers
Tim

timmehtimmeh, 2021-02-15:
Closing due to lack of response - @Rally037 feel free to re-open this if you want to discuss any further.  Cheers!

closed 2021-02-15 by timmehtimmeh

Rally037, 2021-02-15:
Apologies Tim

Working on it, and been away for a bit!

Regards

Jp

timmehtimmeh, 2021-07-21:
For completeness, in case anyone searching Google comes across this issue instead of #75 -

IdeaBuilder 3D40 and 3D45 support have been added to the plugin as of[ version 0.7.0 ](https://github.com/timmehtimmeh/Cura-Dremel-Printer-Plugin/releases/tag/0.7.0) - the plugin is available in the Cura Marketplace.

As always - I welcome suggestions for improvements.

Cheers
Tim