metalman3797/Cura-Dremel-Printer-Plugin #72: error

txtailtocher, 2021-02-20:
For whatever reason im getting a MIME-TYPE error. any suggestions?

txtailtocher, 2021-02-20:
SOLVED: 

sorry just rusty

closed 2021-02-20 by txtailtocher

timmehtimmeh, 2021-02-20:
Hi @txtailtocher - 

I've never seen that error - can you share more information/screenshots.

1. What OS are you running?
2. What Cura version?
3. What are you doing when  the MIME-TYPE error happens?
4. Where does the error appear - on the main cura window or somewhere else?
5. Does this error happen with only a specific item that you're trying to 3d print, or does it happen with any (i.e does it happen with [3D Benchy](https://github.com/CreativeTools/3DBenchy/raw/master/Single-part/3DBenchy.stl)

timmehtimmeh, 2021-02-20:
@txtailtocher - Glad you solved it.  If you care to share how you did it might help others =)

Cheers!
Tim

txtailtocher, 2021-02-20:
I'm honestly not 100% if I did solve it the right way but I found my way around my issue. (Important, I usually don't read instructions) 

I had loaded up a calibration cube to test the plug-in however after slicing I had my SD in the computer so I hit "save to removable drive" as normal; However, I received a "MIME-TYPE" Error, confused by this I did some quick google and found that the error mainly appeared for Unbuntu/Linux users; being I use a windows system for my printing I figured it probably had nothing to do with my OS ( I do a decent amount of CAD so my plugins/directories and so on are managed somewhat responsibly).

I noticed (after my failure haha) in the README that there was mention of the file being saved as a g3drem so instead of saving directly to the SD I opted to save to a file path. Then from the window, I simply selected the SD as the save location and it worked flawlessly.

 Just in case this truly is something new ( I blame my lack of reading Tim's write-up on how to use it for my failure)  I will upload a couple of screenshots showing my errors and the calibration cube. ( I did test it with other files E.G. STL, 3MF, OBJ. without success saving directly to the SD but again by saving via File Path there were no issues whatsoever). 

[image]

In addition to this; I would like to compliment Tim on his work. The plug-in works beautifully. The presets that come with it are nearly perfect and my calibration cube is coming out extremely clean for having only touched the retraction settings. I am writing this real-time so as soon as I pull the cube out I'll take some measurements. photos and continue to upload them as well as recreating the original issue.

That being said Thank you, Tim, for not only being responsive to this post but for making a great bit of software.

txtailtocher, 2021-02-20:
The following photos are of the cube which is 20x20x20 with 30% infill, 3mm retraction at 47mm/s and everything else stock on the (.32 Fast) preset.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

 Along with my error message 
[image]

timmehtimmeh, 2021-02-20:
@txtailtocher  

Thanks for the kind words 😄 and especially thank you for taking the extra time after closing the issue to provide additional details.  I really appreciate it!  👍  

Your dimensions look pretty great for not having modified too many settings - each printer is different so the plugin gets them generally right and then users can tweak them to get closer for their particular printer - looks like your printer is pretty close to the reference printer 😉 

If I could ask two more favors, just in case it's relevant.  Do you mind uploading your 20mm cube model here (you might have to zip the file up).    Also if you don't mind uploading your cura log located at %LOCALAPPDATA%\Roaming\cura\4.8\cura.log I'll take a look and see if the plugin is logging any errors.  To the best of my knowledge that particular error is coming somewhere from within the depths of Cura, but it's possible that the plugin is doing something strange on your system to trigger it.

Regardless of whether you upload or not I'll spend some time this weekend banging on this this weekend to see if I can recreate what you're seeing.   As I said above, I haven't seen this particular message, and haven't experienced any issue saving .g3drem files to either my SD card directly or to my desktop folder in the past.  I'll post here if I discover anything relevant.

And in an ideal world you shouldn't have to read the instructions (but you might discover some of the more obscure features of the plugin if you do ) - so I definitely don't fault you for not reading them 

Cheers

txtailtocher, 2021-02-20:
@timmehtimmeh 

Unfortunately, I cannot upload the calibration cube as it originates from an additional plugin for Cura. However, you can find that plugin at 

https://github.com/5axes/Calibration-Shapes 

Below is my log for Cura 

[cura.log.zip](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/files/6014115/cura.log.zip)

As well, This printer is a workhorse; it has well over 1000 printing hours (still running OEM hardware all the way around surprisingly) and has yet to fail me. That is why I felt the need to mention and show how good the cube looked for the printer being as beat up as it is. 

Good luck and I look forward to your findings! 

Cheers :)

timmehtimmeh, 2021-02-20:
Thanks for log & additional info.  I've done some initial testing using the plugin and writing the calibration cube to my desktop folder, as well as to a USB thumb drive, and a SD card inserted into a USB->SD card reader.  In all three cases I haven't yet seen the mime_type error, but there are a few other things I can try.

I realized that I sent you the wrong directory for the cura logs - sorry =(.  The correct directory is: `%APPDATA%\cura\4.8\cura.log`   (not %LOCALAPPDATA%) I've looked through the log that you uploaded and I can see where the dremel plugin got installed, however I don't see anywhere in your log that indicates that the dremel plugin began writing out a file. I think cura flipped logs right before you started trying to save the file (hence the name `cura.log.1` instead of `cura.log`).  In my first pass at it I don't see anything unusual in the log that you uploaded.

Specifically when the plugin writes out a g3drem file  there will be some lines like this (below I'm writing to the D:\Untitled.g3drem file)
Note that the lines that start with `Dremel3D20.Dremel3D20` come from the plugin and the other lines come from Cura
```
2021-02-19 22:19:32,024 - DEBUG - [MainThread] RemovableDriveOutputDevice.RemovableDriveOutputDevice.requestWrite [85]: Writing to D:/Untitled.g3drem
2021-02-19 22:19:32,159 - INFO - [MainThread] Dremel3D20.Dremel3D20.find_images_with_name [402]: Dremel 3D20 Plugin looking for image with name D:\Untitled.[jpg,gif,bmp]
2021-02-19 22:19:34,070 - DEBUG - [MainThread] Dremel3D20.Dremel3D20.find_images_with_name [416]: Dremel 3D20 Plugin did not find any appropriate image files  - trying to take screenshot instead
2021-02-19 22:19:34,071 - DEBUG - [MainThread] Dremel3D20.Dremel3D20._createSnapshot [154]: Creating thumbnail image...
2021-02-19 22:19:34,076 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
2021-02-19 22:19:34,081 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\overhang.shader]...
2021-02-19 22:19:34,101 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\transparent_object.shader]...
2021-02-19 22:19:34,116 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura 4.8.0\resources\shaders\striped.shader]...
2021-02-19 22:19:34,258 - DEBUG - [MainThread] Dremel3D20.Dremel3D20._createSnapshot [156]: Thumbnail taken
2021-02-19 22:19:34,263 - INFO - [Thread-7] Dremel3D20.Dremel3D20.write [541]: Finished Writing Dremel Header.
2021-02-19 22:19:34,264 - INFO - [Thread-7] Dremel3D20.Dremel3D20.write [551]: Got active build plate
2021-02-19 22:19:34,297 - WARNING - [Thread-7] UM.Decorators.deprecated_function [23]: <function GlobalStack.extruders at 0x000001A157D94488> is deprecated (since 4.4): Please use extruderList instead.
2021-02-19 22:19:34,300 - INFO - [Thread-7] Dremel3D20.Dremel3D20._serialiseSettings [677]: No custom settings found, not writing settings to g-code.
2021-02-19 22:19:34,304 - DEBUG - [Thread-7] UM.FileHandler.WriteFileJob.run [80]: Writing file took 2.149125099182129 seconds
```

The easiest thing to search for in a log is the following line(s):  
`Dremel3D20.Dremel3D20.write`   or `Finished Writing Dremel Header`

This gets logged when the plugin finishes writing the proprietary dremel header, and comes from [this line of code](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/blob/8b1f3c6eb9565c8b1f0ab780f4ba8e4c220459dc/plugins/Dremel3D20/Dremel3D20.py#L541).  If the header writing fails then the log will say ["Error Writing Dremel Header."](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/blob/8b1f3c6eb9565c8b1f0ab780f4ba8e4c220459dc/plugins/Dremel3D20/Dremel3D20.py#L537)

However - even if I don't manage to duplicate your issue this has not been wasted effort.  Looking through the log you sent it appears that the plugin is calling into a deprecated function and I need to make an update.  But other than that the plugin is basically requesting to write to D:\Untitled.g3drem, then taking a screenshot, then writing the g3drem header and finally actually doing the write to the drive.

Next time I turn on my 3D20 I'll look at the number of hours printed on my machine - I bet you've got me beat by a good margin!

I'll look more into this tomorrow and update if I discover anything else.

timmehtimmeh, 2021-02-20:
@txtailtocher  - I've got 851 hours on my printer - as I predicted you got me beat

I think I'm able to duplicate your issue.

Investigating your log a bit more I think I found the line where you're getting the mime_type error

`2021-02-19 16:19:06,299 - WARNING - [MainThread] cura.UI.PrintInformation.setBaseName [366]: Unsupported Mime Type Database file extension TestPart2435533290296`

It appears that something inside of cura is trying to handle a part with the extension `TestPart2435533290296`. 

When I tested with only the following plugins installed I had no errors:
[image]

Looking at your log I see that you've got two additional plugins that I don't have:

1. ArcWelderPlugin
2. CuraBlender

When I installed ArcWelderPlugin and sliced the part I didnt' see the mime_type issue.  However after installing CuraBlender (with ArcWelder, Dremel3D20, and CalibrationShapes also installed) I started getting the mime-type error:
[image]

I'll dig into it some more, but there seems to be something going on after installing the CuraBlender plugin.  If I uninstall that plugin then the "Save to removable drive" option once again works without issue.

txtailtocher, 2021-02-21:
@timmehtimmeh 

Wonderful! It seems your effort has been well rewarded. 

I do find it odd as you found it saving to the D:/ as the SD when plugged in is labeled as E:/ 

Although it may not make a difference I do have a dock plugged in via USB which allows me to insert multiple types of SD as well as others. 

I read your post as mentioned above and it seems like that might be the fix. I have not tried it yet however when I am able to I will attempt it and update you as well. I suppose the worst-case scenario is that I would have to delete and uninstall my plug-ins and reinstall them should they break.

timmehtimmeh, 2021-02-21:
@txtailtocher  - FWIW when I referenced the d: drive that's on my computer :wink: 

Here's my best guess as to what's happening
1. When you click "Save to removable drive" Cura then tries to make a best guess as the file format to save.  
2. [The Dremel3D20 printer definition says that it supports a file format/mime-type of "application/x-g3drem"](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/blob/8b1f3c6eb9565c8b1f0ab780f4ba8e4c220459dc/resources/definitions/Dremel3D20.def.json#L10).  
3. Cura then queries which plugins that can write that file format
4. From what I can tell the CuraBlender plugin doesn't define which mime-types it supports for writeing, and as a result Cura picks that plugin to write out the g3drem format (note -it's entirely possible that I'm completely misguided on this point).
5. The curaBlender code doesn't know how to write a file that the printer can support and the mime-type error pops up.

The log you posted shows multiple mime-type errors, however it doesn't show any of the logs where the Dremel3D20 plugin write function is called.  If it did there would be lines similar to the following in your log:

`Dremel3D20.Dremel3D20.write [541]: Finished Writing Dremel Header.`

Anyway - that's currently my best guess.  As you saw I've posted an [issue on the CuraBlender plugin page](https://github.com/awiegel/CuraBlender/issues/4) Once the author responds hopefully we can get this sorted out.

awiegel, 2021-02-22:
Hey -

I'm the said author of the [CuraBlender](https://github.com/awiegel/CuraBlender) plugin.
Tim's solution with adding an explicit mime type to my plugin's mesh writer works
and the update should be released in the next days.

Cheers and thanks for the effort!
Alex