metalman3797/Cura-Dremel-Printer-Plugin #75: I have a 3D45 for you - drop me a line to jp@lihou.co.uk !  thanks

Rally037, 2021-02-26:


Rally037, 2021-02-26:
Drop me a line - I have a 3D45 for you! sorry about the delay - I missed the thread expiry date

timmehtimmeh, 2021-02-26:
@Rally037  - no worries at all.  I've sent you an email, so be on the lookout and let me know if it doesn't arrive.

FWIW you can always re-open a closed issue with the "Reopen Issue" button
[image]

lenigma1, 2021-03-27:
@Rally037 Your a scholar and a gentleman!
I am humbled by your generosity to the beleaguered 3D45 community!
I'm certain im not the only one with a little buyers remorse (basically unusable with Octoprint? that's not right...), but you contribution to the gifted @timmehtimmeh will no doubt open up a few doors, for this ridiculously sheltered but very promising printer!
My deepest thank you!

thbl, 2021-04-22:
I also have a 3D45 is there an plugin for that model ? ;D

lenigma1, 2021-04-22:
Here.

https://github.com/Xcessity/CuraDremelSupport

For now, this is what we got to play with, and it is helpful.

Although a Dremel 3D45 is being(or has been?) supplied to timmehtimmeh,
it's reasonable to assume he's going to need some more time to unlock the
quagmire that is the 3D45 frankin-hybrid-code on our motherboards.... That
I JUST had replaced, and should be back in my hands in the coming few days!

thbl, 2021-04-22:
@lenigma1 thanks already got that one tho ^^  jsut courious if any one had something cooler yet  ... ;D

lenigma1, 2021-04-22:
I know Tim(for short) is hard at work with his new toy

I too am patiently waiting to see if he can crack Dremels absolutely
bazaar, and broken might I add, usage of http protocol in the file streams
being sent to the printer, that's broken ALL aftermarket software usage for
us...

Yes, technically octoprint does now work, but basically all you can do is
view printing parameters, and START a print.
No pause.
No stop.
Nadda.

Heck, I went a bought a big Tronxy 400mm corexy, had Marlin 2.0 and
octoprint up and running in the first day, including the kit thats build
from scratch, and even TronXY's broken ChiTu firmware accepts basic M
commands, and gcode input.
Alas, even my extensive knowledge with modding Pronterface couldn't get
basic commands to work on the 3D45.
Wait.
That's a lie, I remember being able to again, just start a print... But I
couldn't get the leveling sensor servo to activate to bring the sensor back
up... So had I let it continue, it would have printed, sensor down, lol.

*Good things come, to those who wait*

Cheers

timmehtimmeh, 2021-04-22:
@lenigma1  @thbl 

No updates yet -

>I too am patiently waiting to see if he can crack Dremels absolutely
bazaar, and broken might I add, usage of http protocol in the file streams
being sent to the printer, that's broken ALL aftermarket software usage for
us...

Please understand that any initial implementation of the 3D45 will be very similar to the current 3D20 implementation - it will allow the user to install the necessary printer files & material definitions via the plugin and it will generate a .g3drem file that will be exported to a USB drive.   Please see the discussion [here](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/issues/71) on what will be initially supported and what won't be. 

Cheers
Tim

lenigma1, 2021-04-24:
I was not aware the 3D45 used, or printed with its little brothers proprietary .g3drem files!

I've always used gcode and .curamaterials(I think it's called...) And never considered it's usage in any way, lol!

The very fact you are going to focus on profile settings and tuning ~ which cannot be done very well in its stock iteration... I've got some GHASTLY pro ts to prove it ~ sends my printer soul soaring!

THANK YOU!

Although my 3D45 came back today from a 7 week stint @ the Dremel shop replacing the main board(which was faulty from day 1... My very first bed level ever failed due to z failure), I now have hope to cling to, as ABS, REAL ABS, not Dremels modified PLA "Eco-ABS", NEVER printed well, dispite dozens and dozens of settings changes and test prints!

TL;DR - I'm ecstatic with any effort being put into this expensive old girl!

I'm fully aware Octoprint functionality, if even possible, is WAY down the road.

Also, I'd be very happy to alpha/beta/control test, if needed.

timmehtimmeh, 2021-04-24:
@lenigma1  - 👍  If / When I have an update for the 3D45 I'll post back here.  

FWIW - I was able to get octoprint working on my 3D20 - I wrote up instructions on how I did [here](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/tree/stable/docs/octoprint).

lenigma1, 2021-04-24:
Awesome, thanks!

timmehtimmeh, 2021-07-16:
At long last I'm happy to announce that the new "Dremel Printer Plugin" is now in the Cura marketplace.  Huge thanks to @Rally037! 

[image]

I've removed the older 3D20 plugin from the marketplace, as this plugin supersedes it.  This plugin will detect if the old 3D20 plugin is installed and will prompt you to uninstall the older 3D20 plugin.  

Please note that I haven't tested the 3D40 settings for any of the Dremel materials, as I don't have access to a 3D40.  I'll continue tweaking the per-printer-per-material-per-quality settings for the Dremel materials for the 3D20 and 3D45.  I'd welcome help or recommendations on any improvements to the quality settings for the Dremel specific materials for any of the printers.  After that I'll start poking around the camera.

@lenigma1 - I'm not sure if you've seen it, but there's now a [plugin that allows you to upload to Dremel Printers](https://github.com/lwerner-lshigh/Dremel3D-CuraPrintr).  Eventually I'd like to include similar functionality in my plugin, but since there's already a plugin out there that does this I'd like to work on things that users can't do at this moment.

closed 2021-07-16 by timmehtimmeh