metalman3797/Cura-Dremel-Printer-Plugin #77: No Dremel printer available in printers list

polybius80, 2021-06-16:
Hi I installed your plug-in in various cura versions but the dremel printer is not listed, I tested many versions of cura and your last 2 versions of the plug-in and also installed manually in each case and restarted cura many times, there is the menu of the plug in, its installed but still not printer in the list

I am using the latest ubuntu (21.04) as reference 

thank you for your time

timmehtimmeh, 2021-06-16:
Hi @polybius80 

I just tested using Ubuntu 20.10 following steps outlined below and was able to add a Dremel 3D20 from a clean install of Cura.

**Steps**
1)  Download [Ultimaker Cura 4.9.1 AppImage](https://software.ultimaker.com/cura/4.9_20210511_Ultimaker_Cura-4.9.1.AppImage)
2) Moved the .appimage to my Desktop and ran `chmod 755` on the Appimage
3)  Reset Cura to the factory state by deleting cura's caches (done by running the following commands):
```
rm -rf ~/.local/share/cura/*
rm -rf ~/.config/cura/* 
```
3. Launch Cura & stepped through the initial prompts.  When it came time to choose a printer you can see that the Dremel 3D20 wasn't an option.  I simply selected the Ultimaker S5 default option
[image]

4. Open up Cura's marketplace (note that because I deleted Cura's caches I have no plugins installed.
[image]

5. Navigate to the Marketplace's plugins directory, find the Dremel 3D20 plugin (ver 0.6.4) and click the Install Button
[image]

6. At this point I was prompted to restart Cura, so I clicked the "Quit Ultimaker Cura" button
[image]

7. I then re-launched the Appimage, navigated to `Settings->Printer->Add Printer`
[image]

8. Scrolling down the Dremel printer is now an option.
[image]

Please try the above sequence...if following those steps don't work then if you don't mind uploading your cura.log file located in
`~/.local/share/cura/4.9/cura.log` it would be helpful.   Additionally if the above sequence doesn't work please let me know if you see a Dremel3D20.def.json in `~/.local/share/cura/4.9/definitions/Dremel3D20.def.json`

timmehtimmeh, 2021-06-17:
One additional note for what it's worth: I just upgraded my computer to Ubuntu 21.04 and followed the above steps again (once again clearing cura's cache first to start cura from default settings) and was also able to successfully install the 3D20.

polybius80, 2021-06-17:
it now works, thank you

apparently it was something with the cura cache

timmehtimmeh, 2021-06-17:
Awesome - glad to hear that it's working now.   :partying_face:  

If you'd ever used a **really old** version of the plugin and toggled the the plugin's preferences page->install plugin files checkbox that _may_ have caused the issue, I'll review the code to see if there's a chance for improvement in it.

Cheers!

closed 2021-06-17 by timmehtimmeh