metalman3797/Cura-Dremel-Printer-Plugin #78: cura 4.91 not loading dremel plugin

Ant628, 2021-06-23:
i added the plugin to cura 4.91 and restarted but the printer is not added to the list. is there a step that i am missing?
thanks 
tony

timmehtimmeh, 2021-06-23:
Hi @Ant628 -

Thanks for the report - you're not missing anything - you're the second person in a week to report a similar issue, making me think that Ultimaker either has a bug, or changed something behind the scenes, but I've been unable to duplicate the issue on either windows or Linux. See https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/issues/77 for the solution that worked for that user on Linux

What OS are you running?  Can you upload your cura.log for me to take a look at?  It's located at:

**Windows:** `%APPDATA%\cura\4.9\cura.log`
**MacOS:** `$USER/Library/Application Support/cura/4.9/cura.log` 
**Linux:** `$USER/.local/share/cura/4.9/cura.log`

Ant628, 2021-06-23:
[cura.log](https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/files/6704660/cura.log)

here is the cura log. i have windows 10 pro version 2004 os build 19041.572

thanks for the help

timmehtimmeh, 2021-06-24:
Thanks for the log @Ant628 -

From it, it appears that you at least had the Dremel3D20 plugin installed at one time, as I can see some notes about some of the files that the plugin installs, but the last reference I see to those files was made on June 8.  I don't see any references to the files that this plugin installs in the logs from June 22 or June 23.  

From the log appears that right now you don't have the plugin installed.  Is that possible?   You have a [dremel3D_upload plugin installed](https://github.com/lwerner-lshigh/Dremel3D-CuraPrintr), but that is a different plugin than this one.  I see this line in your log
`2021-06-23 08:49:04,221 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin dremel3d_upload`

but I do not see a line similar to this, which would indicate that the Dremel3D20 plugin is installed.
`2021-06-23 17:06:24,143 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin Dremel3D20`

If you go to cura->extensions do you see the "Dremel 3D20 Printer Plugin" as an option?
[image]

Also - do you see "Dremel 3D20" plugin installed if you go to Marketplace->Installed?
[image]

If not - go to the Marketplace->Plugins and install the "Dremel3D20" plugin. 

Cheers!

Ant628, 2021-06-24:
i was so excited to see a dremel plug in that i chose the dremel upload instead of the 3d20.  its good. totally my fault. 
going to start playing now. 
thanks for the fast reply and help.

timmehtimmeh, 2021-06-24:
I hadn't seen the "Dremel Upload" plugin until yesterday either - looks like it came out only a few weeks ago, so thanks for bringing that to my attention.

I'm glad it was an easy fix 😄

closed 2021-06-24 by timmehtimmeh