metalman3797/Cura-Dremel-Printer-Plugin #83: Remote monitoring camera 3d45

takacsa, 2021-12-30:
Is there any way to connect the built-in camera on the 3d45 so you can monitor it remotely?  The Dremel slicer allows for it, I figure there has to be a way to do it through Cura.

timmehtimmeh, 2021-12-30:
Hi @takacsa 

Great question - it's on my list of upgrades that I'd like to implement, but right now the answer is that receiving data from the camera is not directly implemented in my plugin yet.

I'll keep this issue open and will update it if/when I make any progress toward this end,  but please understand that I'm extremely busy with work these days and progress toward this will be extremely slow at best.

If you've got any technical/coding skills I'd be happy to have you make a pull request if you got an implementation working -

Cheers
Tim

takacsa, 2021-12-30:
I would if I could but the last time I even attempted to code anything, it was in line number BASIC, Pascal, or HTML 4 back ~ <'95.

lenigma1, 2022-02-14:
There MUST be a way of grabbing that data stream - & I say this simply due to Dremels packaged 3DPrinterOS.
I have on many occasions set  up my 3D45 (IDK if 3DPrinterOS is packaged with the 3D20 OR 3D40??), with whatever filament, bed level, etc, then left for a clients, 3D files in tow,  and once the client was happy with the final touches, upload to 3DPrinterOS, slice n print.
I always enjoyed watching the wow factor go across nearly every clients face when they realized I had in fact started their print, AND they could watch it's humble beginnings.

I just don't know enough (yet) about coding to be of any use, beyond an alpha/beta tester.
I know that video stream is leaving my home network, and through 3DPrinterOS, I'm able to view it ANYWHERE there's internet and a secure connection to said OS.

I'm certain the wiser folk here already knew this... I just figured I should drop my 2 cents, in case it jogs someone's memory the right way, lol!

Cheers!😁

timmehtimmeh, 2022-02-14:
@lenigma1 - I don't think that grabbing the stream (at least not from a machine on the same local network as the printer) is the biggest challenge - the 3D45 has a small internal web server that the MJPEG stream (and jpg snapshots) can be accessed from ( **http://_dremel printer address_:10123/stream_simple.html** )  the Dremel firmware seems to be using a (mostly unmodified?) version of [MJPEG Streamer](https://sourceforge.net/projects/mjpg-streamer/) as is evidenced by the default web page

The more challenging part of this is related to the fact that from what I can tell Cura doesn't include/distribute the QT multimedia modules that would make decoding mjpeg streams cross platform & much easier to integrate.  They may have chosen to not include the QT multimedia support for a number of different reasons (i.e. security, distributable size, etc....), but without them distributed with Cura an alternative would have to be found.  I spent a bit of time looking for python based stream decoders without much success, and have thought about looking at using libvlc or another library to decode the mjpeg stream, but I haven't gotten too far down that path.  As I said above work is keeping me pretty busy at the moment.

Regardless - before I ever get video streaming working I plan on working on grabbing the snapshot images working where the plugin will get the static image from **http://_dremel printer address_:10123//?action=snapshot** and update it periodically (ideally this would feed into an interface to make a time-lapse video, but one step at a time...)

Cheers
Tim

timmehtimmeh, 2022-02-15:
More notes to myself: 

the following code (hacked together & not production ready) can grab and decode an image (images?) from a MJPEG streamer server and I believe uses QT libraries that are included in Cura (although that needs further testing).

[image]

```python
import urllib.request
from PyQt5.QtGui import QImage, QPixmap
from PyQt5 import QtWidgets, QtCore
import sys

class MainWindow(QtWidgets.QMainWindow):
    def __init__(self):
        super(MainWindow, self).__init__()
        self.img = QImage()
        self.label_imageDisplay = QtWidgets.QLabel()
        self.label_imageDisplay.setAlignment(QtCore.Qt.AlignCenter)
        self.label_imageDisplay.setScaledContents(True)
        self.label_imageDisplay.setMinimumSize(1,1)

        self.startbutton = QtWidgets.QPushButton("Start")
        self.startbutton.clicked.connect(self.get_next_frame)

        self.stopbutton = QtWidgets.QPushButton("Stop")
        self.stopbutton.clicked.connect(self.Stop)

        self.ipAddr = "192.168.XXX.XXX"
        self.port = "10123"

        vlayout = QtWidgets.QVBoxLayout()
        hlayout = QtWidgets.QHBoxLayout()
        hlayout.addWidget(self.startbutton)

        hlayout.addWidget(self.stopbutton)
        vlayout.addLayout(hlayout)

        vlayout.addWidget(self.label_imageDisplay)
        widget = QtWidgets.QWidget()

        widget.setLayout(vlayout)
        self.setCentralWidget(widget)
        self.stopping = False
        self.bytes = bytes()

    def Stop():
        self.stopping = True
        
    def get_next_frame(self):
        stream_url = 'http://'+self.ipAddr+':'+self.port+'/?action=stream'
        stream = urllib.request.urlopen(stream_url)
        while not self.stopping:
            QtWidgets.QApplication.processEvents()
            self.bytes += stream.read(1024)
            a = self.bytes.find(b'\xff\xd8')
            b = self.bytes.find(b'\xff\xd9')
            if a != -1 and b != -1:
                jpg = self.bytes[a:b+2]
                bytes = self.bytes[b+2:]
                img = QImage()
                if(img.loadFromData(jpg, "JPG")):
                    pixmap01 = QPixmap.fromImage(img)
                    self.label_imageDisplay.setPixmap(pixmap01)
                else:
                    print("ERROR")

if __name__ == "__main__":
    app = QtWidgets.QApplication(sys.argv)
    window = MainWindow()
    window.resize(640, 480)
    window.show()
    sys.exit(app.exec_())
```

lenigma1, 2022-02-16:
Thank you for you wonderfully derailed and prompt reply!

I was aware, and use virtually every time the Dremel prints, of the local address that allows viewing of current print through the Internal camera.
I guess I forgot to include that in my previous venting of hot air😅😋😁

I will continue to poke around in subreddits and various forums, but not I'll include python stream decoders as well, see if I can just once offer this 3D45 dev community a bit of bright light!

Stay safe, & cheers!

- I'll also keep an eye out around here, Incase you in fact to need someone to test something

timmehtimmeh, 2022-02-16:
As I mention in my edit the code above grabbed multiple frames from the stream, but after displaying the first one it wouldn't update with subsequent data.  The following code rectifies this by using a QT signal to emit the newly grabbed frame as an image.  It's based on a modified version of [this stack overflow answer][(https://stackoverflow.com/questions/44404349/pyqt-showing-video-stream-from-opencv)](https://stackoverflow.com/a/44404713)  

[image]
[image]

I still need to check to see if something like this would work within the Cura plugin architecture but I'm hopeful that it can.  If so then the code needs a bit of cleanup as I see a few potential issues that I'd like to eliminate from the code below before including it in the plugin.  I'll look at this more over the weekend.

```python
import urllib.request
import sys
from PyQt5.QtWidgets import  QWidget, QLabel, QApplication
from PyQt5.QtCore import QThread, Qt, pyqtSignal, pyqtSlot
from PyQt5.QtGui import QImage, QPixmap

class Thread(QThread):
    changePixmap = pyqtSignal(QImage)

    def run(self):
        ipAddr = "192.168.XXX.XXX"
        port = "10123"
        stream_url = 'http://'+ipAddr+':'+port+'/?action=stream'
        stream = urllib.request.urlopen(stream_url)
        abytes =  bytes()
        img = QImage()
        while True:
            abytes += stream.read(1024)
            a = abytes.find(b'\xff\xd8')
            b = abytes.find(b'\xff\xd9')
            if a != -1 and b != -1:
                jpg = abytes[a:b+2]
                abytes = abytes[b+2:]
                if(img.loadFromData(jpg, "JPG")):
                    self.changePixmap.emit(img)
                    
class App(QWidget):
    def __init__(self):
        super().__init__()
        self.title = "Stream"
        self.initUI()

    @pyqtSlot(QImage)
    def setImage(self, image):
        self.label.setPixmap(QPixmap.fromImage(image))

    def initUI(self):
        self.setWindowTitle(self.title)
        # create a label
        self.label = QLabel(self)
        self.label.resize(640, 480)
        th = Thread(self)
        th.changePixmap.connect(self.setImage)
        th.start()
        self.show()

if __name__ == "__main__":
    app = QApplication(sys.argv)
    window = App()
    window.resize(640, 480)
    window.show()
    sys.exit(app.exec_())
```

timmehtimmeh, 2022-02-21:
After a number of false starts I've finally gotten the qt QQuickImageProvider class working with PyQt to update a simple QtQuick UI.  I'm hopeful that 

I looked at how Cura displays their camera images, and started by trying to integrate into that, but they display their images on the "monitor" tab, and integrating into that would involve a number of changes to the plugin that I don't have the time to do right now.

I'm hopeful that between either the code above of the code below I'll be able to add a menu item to make a separate camera popup window that will display the camera from the 3D printer

I'm pasting this code below (and attaching it as a zip) because It was annoying enough trying to find working pyqt examples for the QQuickImageProvider that updated the image that I want it to be indexed by search engines to help anyone else who's looking:

**main.py**
```python3
from PyQt5.QtGui import QGuiApplication, QImage
from PyQt5.QtCore import QUrl, pyqtSignal, QObject, QThread
from PyQt5.QtQuick import QQuickView
from DremelCameraImageProvider import DremelCameraImageProvider
import urllib.request
import sys

class CameraGrabThread(QThread):
    updateImage = pyqtSignal()
    running = False

    def setImageProvider(self, imgProvider: DremelCameraImageProvider):
        self.dremelImageProvider = imgProvider

    def setIPAddress(self,ip):
        self.ipAddr = ip

    def stop():
        self.running = False
    
    # TODO - make this more robust 
    def start():
        stop()
        run()

    def run(self):
        # if we're already running just return
        if self.running:
            return

        self.running = True
        port = "10123"
        stream_url = 'http://'+self.ipAddr+':'+port+'/?action=stream'
        stream = urllib.request.urlopen(stream_url)
        abytes =  bytes()
        img = QImage()
        while self.running:
            abytes += stream.read(1024)
            a = abytes.find(b'\xff\xd8')
            b = abytes.find(b'\xff\xd9')
            if a != -1 and b != -1:
                jpg = abytes[a:b+2]
                abytes = abytes[b+2:]
                if(img.loadFromData(jpg, "JPG")):
                    self.dremelImageProvider.setImage(img)
                    self.updateImage.emit()
            if abytes > 5000000:
                self.stop()  # stops the thread
                self.start()

class FrameTriggerClass(QObject):
    # signal that gets sent to the DremelCameraImage.qml to trigger
    # a reload of the image (the image comes from DremelCameraImageProvider)
    signalNewFrameReady = pyqtSignal(int,arguments=['currFrameNum'])
    
    def __init__(self):
        super().__init__()
        self.nextFrameNumber = 0

    def slotNewFrameReady(self):
        self.nextFrameNumber += 1
        self.signalNewFrameReady.emit(self.nextFrameNumber)

if __name__ == "__main__":
    app = QGuiApplication([])
    app.setApplicationName("Dremel Camera Viewer")

    # create the image provider
    imageProvider = DremelCameraImageProvider()

    # create a helper to trigger the image provider
    # to update images when new frames come in
    frameTrigger = FrameTriggerClass()

    # create a camera viewer UI
    viewer = QQuickView()
    viewer.rootContext().engine().addImageProvider("DremelCameraImageProvider",imageProvider)
    viewer.rootContext().setContextProperty("newimagetrigger",frameTrigger)
    viewer.setSource(QUrl("DremelCameraImage.qml"))
    viewer.show()

    # create the thread that grabs camera images
    th = CameraGrabThread()
    th.setImageProvider(imageProvider)
    th.updateImage.connect(frameTrigger.slotNewFrameReady)

    # set the IP address here
    th.setIPAddress("192.168.XXX.XXX")
    th.start()

    sys.exit(app.exec_())
```
**DremelCameraImageProvider.py**:
```python3
from PyQt5.QtCore import pyqtSlot,QSize
from PyQt5.QtQuick import QQuickImageProvider
from PyQt5.QtGui import QImage, QColor

class DremelCameraImageProvider(QQuickImageProvider):
    
    def __init__(self):
        super().__init__(QQuickImageProvider.Image)
        self.cameraImage = QImage(QSize(640,480), QImage.Format.Format_RGBA8888)
    
    # sets the camera image to send to the UI
    def setImage(self, newImage: QImage):
        self.cameraImage = newImage

    # returns the latest camera mage
    def requestImage(self,  id_, requestedSize):
        return self.cameraImage, self.cameraImage.size()
```

**DremelCameraImage.qml**:
```python3
import QtQuick 2.1
import QtQuick.Controls 1.0

Rectangle {
    width: 800
    height: 600

    Column {
        anchors.centerIn: parent
        Image { 
                id: cameraImage
                width: 800
                height: 600
                source: "image://DremelCameraImageProvider/" + 0

                function reload(currFrameNum) {
                    //console.log("Curr# is ", currFrameNum);
                    source = "image://DremelCameraImageProvider/" + currFrameNum
                }
        }
    }

    Connections{
            target: newimagetrigger
            function onSignalNewFrameReady(currFrameNum)
            {
                cameraImage.reload(currFrameNum)
            }
    }
}
```

[DremelCameraImage.zip](https://github.com/timmehtimmeh/Cura-Dremel-Printer-Plugin/files/8105765/DremelCameraImage.zip)

timmehtimmeh, 2022-02-21:
[I've made some progress.](https://github.com/timmehtimmeh/Cura-Dremel-Printer-Plugin/commit/83158eee9cae0c57fcaa11204282829c19fd5449) 📷 

I've added a preference setting to enable the user to input the Printer's IP address, 
[image]

and a "View Camera" option to the menu
[image]

Which brings up the camera view in a separate window.  
[image]

The code still to be tested some more, and I need to do some cleanup before it will can be released, but it's progress toward that end

timmehtimmeh, 2022-03-06:
@takacsa  and @lenigma1 

At the bottom of this post is a preview copy of version 0.7.2 of the plugin that enables the 3D45 camera to be streamed in a separate window.  Feel free to test it and let me know if you find any issues.  I've tested it a bunch of different ways and think that I've squashed a majority of the bugs, but having other people test it would be useful before I release it broadly would be helpful 👍 

I'll keep testing it for another week or so, and if I don't find any major issues I'll submit it to Ultimaker for review to be added to the marketplace.

Instructions on how to use the new camera features can be found [here](https://github.com/timmehtimmeh/Cura-Dremel-Printer-Plugin/tree/develop#Dremel_3D45_Camera)

To test the update, please download [Cura-Dremel-Plugin-0.7.2.zip](https://github.com/timmehtimmeh/Cura-Dremel-Printer-Plugin/files/8193535/Cura-Dremel-Plugin-0.7.2.zip) and extract it.  Then drag the `Cura-Dremel-Plugin-0.7.2.curapackage` file onto a running version of Cura.  You should see a message similar to the following:
[image]

Then restart Cura.  

You can validate that the plugin was updated by looking at the version number in the Plugin's menu.
[image]

Cheers
Tim

lenigma1, 2022-03-06:
And when you say "drag it on to the running package" do you mean cura's installation folder? Or the actual exe?

lenigma1, 2022-03-06:
Crap, hit send too fast -> THANK YOU EVER SO MUCH, for all you do to break down these walls to Dremels closed off garden!
Cheers my friend, I'll be on this tomorrow!
...after your response of course, lol 😅

timmehtimmeh, 2022-03-07:
>And when you say "drag it on to the running package" do you mean cura's installation folder? Or the actual exe?

Start cura running, and drag and drop the .curapackage onto the main cura window

takacsa, 2022-03-07:
Thanks both of you for your work on this.  I installed and ran it.  The webcam works.  It's a slow frame rate and laggy and seems to disconnect frequently - not sure if you can change that - but at least I don't have to walk across the room to see it.
[image]

On a related note, I chopped up my 3d45 to fit 3rd party spools inside the enclosure.  not sure if either of you would be interested in pics or a write-up on that.  Basically get at the Digilab with a Dremel tool.  Ironic, I know.

timmehtimmeh, 2022-03-07:
@takacsa 

Thanks for testing - I really appreciate the time you're taking to help test & report back 👍 

>The webcam works. It's a slow frame rate and laggy and seems to disconnect frequently - not sure if you can change that - but at least I don't have to walk across the room to see it.

The way the plugin works is that it will connect to the camera stream at **http:// dremel printer ip address:10123/?action=stream** ie. http://192.168.255.255:10123/?action=stream (replacing the IP address with the IP of your printer) and parse the image out of the stream and display it.  If the plugin doesn't detect an image for 2 seconds then the plugin thinks that the network connection is disconnected and tries to reconnect.  This may happen for a few different reasons:

1. The network connection between the printer and the computer is slow/having problems
2. The computer is busy doing something and can't get around to processing the new frames that are coming in
3.  something else?  

There are a few questions that I have that may help point in a direction

1.  When you navigate to the camera's stream and view it with a web browser is that also slow/laggy ?
2.  Does your computer's CPU usage seem high when viewing the camera using the plugin's camera viewer?

Regarding using 3rd party filaments - I printed [this external spool holder](https://www.thingiverse.com/thing:3322829)  for third party filaments...seems to work pretty well, but

takacsa, 2022-03-08:
Since my last comment, I replaced my wifi router.  The feed seems to be much smoother now.  I'm not 100% sure the router swap did the trick or if it's just coincidence but I'll keep an eye on it.

Yeah, I was doing the external spool for a while but hated it.  I got the Dremel in part because the enclosure.  Having the spool externally takes more space on your table top, doesn't protect or warm the filament.

As for hacks, the hacks I'm interested in now are 1) Can we override the 270C max heat setting?  And go for like 300+ C?  If not, what parts to order to upgrade it?  2) Can the 720p camera be upgraded with a 1080p or 4k webcam?

timmehtimmeh, 2022-03-09:
Thanks for the update  - I'm glad to hear that the camera feed is smoother with the changes you made to your network.  This weekend I'll try to find a spot that's in the "goldilocks zone" of my wifi coverage that the Dremel can still connect, but far enough away that the connection is poor, and I'll see if I can make any code changes to improve anything.  

Regarding hacks to the 3D45, I'm probably not the right person to ask, as I'm specifically trying to keep mine as stock as possible in order to ensure that the plugin works with the stock printer in a stock configuration.  That said I did successfully flash my 3D20 firmware to Marlin using the code at [this repository](https://github.com/moonglow/FlashForge_Marlin), but I then flashed back in order to be able to test the 3D20 in it's stock configuration. The author of that firmware didn't seem to think that the 3D45 would be able to be flashed in the same way as the 3D20, but you might reach out to them and see if they know any techniques.  From corporate experience I know that the engineering team at Dremel would have had to implement some safety checks to ensure that Dremel was protected legally from people doing stupid things and then suing.   

I've also created [instructions how to get octoprint enabled with the 3D20](https://github.com/timmehtimmeh/Cura-Dremel-Printer-Plugin/blob/stable/docs/octoprint/README.md).  I haven't tested those with the 3D45, but my guess is that it won't work with the 45...may be worth trying and reporting back...

timmehtimmeh, 2022-03-12:
I've been unable to reproduce any laggy/slow connection issues with my printer - I found the spot in my house where the 3D45 won't connect to the wifi, and then if I move it a few inches closer the printer will connect.  However once the printer is connected it seems to maintain a good connection to the wifi, even if I move it back outside the range that it will connect in while the printer is on/streaming.  I also tried keeping the printer in a fixed location and walking my laptop around the front and back yard, but apparently I've done too good a job at eliminating dead wifi spots in my house 😂 

@lenigma1 - have you had a chance to test the plugin?

takacsa, 2022-03-22:
So far so good with the remote monitoring from the same LAN.  Is there a way to view it remotely when not on the same LAN?  Would that involve some router configuration?

***@***.***
ANDRE M. TAKACS, MBA
(913) 240-0200
https://www.linkedin.com/in/andre-takacs/

timmehtimmeh, 2022-03-22:
@takacsa 
>So far so good with the remote monitoring from the same LAN. 

Thanks for continuing to test 😃 I really appreciate it.  I've been busy at work, but have continued testing when I could and so far haven't found any issues still.  

I still haven't uploaded this version to Ultimaker, but will do so soon (maybe a week or two) unless I find a breaking bug.

> Is there a way to view it remotely when not on the same LAN?  Would that involve some router configuration?

I believe so - the printer's camera streams data out on port 10123.  Your router should have some settings to forward requests from your external IP address port 10123 to your dremel printer, at which point you should be able to access the stream from the external IP address.  It should be called "port forwarding" or something similar in your router settings.

That said I haven't tested this configuration and you should be aware that **port forwarding does come with security implications and you are allowing anyone on the internet to access your dremel's camera stream.**  There is a chance that they could find a way to do something more nefarious to your printer and as such I recommend that you don't open up ports in your router unless you understand and accept the reduced security.

See [these links](https://www.google.com/search?client=firefox-b-1-d&q=security+implications+of+port+forwarding) for some more information.

closed 2022-04-23 by timmehtimmeh

lenigma1, 2022-04-24:
I'm so very happy you have found the time to dev for our anti-dev machine!👍👌🤟🖖

TL;DR ~~~~~~~~~~
I was heavily divided time wise the past 6~8 weeks, but it's now all cleared and sorted.
Promise to be a far more engaged tester, and an humbled with your inclusion of us in your testing section!
Anything additional I can help with??
~~~~~~~~~~~~~~~

Now, the long winded version I didn't realize had gotten SO long till I proofread!🤭🙃😋

1st off, I'd like to apologize for my lack of input recently.🙏
My mother passed last month, and with my sister having 3 pre teen kids, & me still on crutches, well, long story short, settling mom's estate took a lot more time and much more effort than anticipated 😅😁

Believe it or not, her home is situated on one of the last areas in the developed world that does NOT have cell service of any kind!🤯
Oh, and remember how terrible and SLOW dial up internet was?!?
Yeah... I got a healthy reminder of precisely and nearly useless dial up actually is while out there as well!🙃😋

However,
I did infact test the plug-in several times, and was actually able - after a fair bit of tinkering with my networks NAS setup - to remotely view the camera from nearly 200kms away! 

Not that anyone will ever be doing that on a regular basis, but nice to know it is possible, and does work.
Unfortunately, I was not able to measure latency in any meaningful way.
Another time, in the near future, I'll have to set up synced watches, and have it focused & physically close to the camera. 

Rest assured though, had I encountered a bug or unsurmountable issue, I would definitely found time to explain such in one of my very few and highly organized (& far too short I should add😅) trips into town to use the local coffee house wifi to preform all manner of business stuff. 

EDIT: As embarrassing as this is to admit, I see here I did infact write out an update to a direct question you asked several weeks back, wondering if I could test the new edition! 🤭☺️.
This is terribly not like me, to not finish up and hit send, and for that, want to offer both my deepest apologies, and a promise never to be that seriously fragmented again, to the point of forgetting something as important as a progress update on software testing! 🙃😉

Mom's estate has been dealt with, and I'm back home in Toronto, ready to upgrade the TronXY with a DUET 3 mini 5+ & matching 7i, and look into this new companies' direction that Dremel corporate sold the Digimaker line too... Cause I believe I was sent what appears to be an application for hardware and software testing! 

Lastly, I am humbled that you would offer up the 2 if us a spot in your tester/ing credits🥲
From here on in, I'ma make good on that accolade you have bestowed upon us!

And congratulations on getting another shiney plugin officially recognized by Cure/Ultimaker!!

So, is there currently anything you need a hand with?
Or just keep driving the plug-in for now, looking to iron out any additional bugs, UF any still exist? 

Stay Safe and Sincerely

Bo Pedersen

takacsa, 2022-04-24:
Great work!  I saw the Cura 5 beta come out and was about to give it a try but I had a feeling the plug in might not work smoothly. I'm looking forward to 5 and updated driver. Bo, how did you view your camera remotely?  I was able to capture it in OBS and stream it through to Twitch.tv/trollwerks  I'd love to find another way though.  Looking forward to Cura 5 & a working driver.  Bo, how did you view your camera remotely?  I was able to capture it in OBS and stream it through to Twitch.tv/trollwerks  Bo, how did you view your camera remotely?  I was able to capture it in OBS and stream it through to Twitch.tv/trollwerks  I'd love to find another way though.  Also, I'm interested in seeing if 4k rez &/or multiple cameras are possible.

Sent from my Verizon, Samsung Galaxy smartphone

timmehtimmeh, 2022-04-24:
Thank you both for your help with encouragement & testing - I really do appreciate it!

Right now I have nothing that i need to test - if either of you have a 3D40 and have better print profiles for the various materials than what this plugin includes that would be useful - I have a 3D20 and a 3D45, but not a 3D40, so I can't test the print profiles on a 3D40, but other than that I can't think of anything else.  I need to start working on the Cura 5 update to the plugin, but I haven't begun that yet (maybe later today, maybe next weekend)

@lenigma1  - I figured that you had tested the 0,7.2 "beta" and since I didn't hear anything back that you hadn't encountered anything that was terribly broken.  I'm happy to hear that was the case.  The only thing that changed from the beta to the final release was a couple of minor documentation edits, so the PDF included when you click `Extensions->Dremel Printer Plugin->Help` is very slightly different in the final released version, but the actual plugin code is the same.

@takacsa  - I hadn't considered OBS as an option to view the video remotely - that's awesomely clever 👍   I'll let @lenigma1 respond on how they got it to work, but as I mentioned earlier your router should have a port forwarding option that you can use to poke a hole in your router's firewall and access the stream directly (same security warnings that I mentioned above still apply) 

FWIW Cura will import the Dremel printer profiles from Ver 4 into Version 5, so as long as your 3D20, 3D40, or 3D45 has firmware to accept gcode files you can give Cura 5 a try without the plugin (Cura will write out .gcode files and not .g3drem files).  Users who don't have Cura 4 & the plugin installed won't be able to add a Dremel printer, but you both wouldn't be affected by that.

Finally @lenigma1  - my condolences for your loss 😢

takacsa, 2022-10-11:
Unrelated 3D45 questions:

I heard the 3D45 doesn’t take the temperature settings from CURA and instead uses what you input directly to the machine when you change the filament.  Is this correct?

Also along the same lines, if I change the nozzle size, does 3D45 read it straight from the settings in Cura or is there a different way to do this?  I bought a 0.4, 0.6, and 0.8mm nozzle for printing faster and using thicker CF-enhanced nylons, etc.

***@***.***
ANDRE M. TAKACS, MBA
(913) 240-0200
https://www.linkedin.com/in/andre-takacs/