metalman3797/Cura-Dremel-Printer-Plugin #87: Network issues 3D45

Rally037, 2022-04-26:
Hi Tim, dropped you an email.  Cura 4.10, OS Big Sur.  I can see the camera of a networked 3D45 by entering the IP address but Cura won't connect to the printer (same IP address - double checked).  Let me know if you need anything more

timmehtimmeh, 2022-04-26:
Hi @Rally037 -

Currently this plugin doesn't support a full-blown network connection to the printers (for sending & receiving files, remote monitoring etc...).  The ver 0.7.2 update to this plugin enables the 3D45 camera stream to be viewed in a separate window, but not more than that.  On my *someday* list is enabling a full network connection to the printers, however that functionality is not yet implemented. 

If you want to send files to the printer via the IP address then there's a secondary plugin called "Dremel 3D upload" in the marketplace which enables you to send files remotely. 

[image]

As I said, eventually I'd like to incorporate that functionality into this plugin, but it's not there yet.

lwerner-lshigh, 2022-05-19:
@timmehtimmeh I'm down to make a pull request!

timmehtimmeh, 2022-05-19:
@lwerner-lshigh - awesome - working with you to combining the functionality of our two plugins has been on my to-do list.

Eventually I'd like to have the 3D40 & 3D45 printers completely integrated into Cura's network discovery workflow via the printer definitions (i.e. `"supports_network_connection": true,` like the [Ultimaker S5](https://github.com/Ultimaker/Cura/blob/64b29779ca808119458eb52d0add91a87c9a25b6/resources/definitions/ultimaker_s5.def.json#L30) ).  That should make it easier to monitor the camera stream via the "monitor" tab in Cura instead of me having to create and manage the window & threads that displays the camera stream myself, but that's probably a few steps down the road.

If you do make a PR, please do so onto the [develop branch](https://github.com/timmehtimmeh/Cura-Dremel-Printer-Plugin/tree/develop) of this plugin.  I try to keep the main "stable" branch concurrent with the latest releases and do any development on the "develop" branch.

Also - I believe you had access to a 3D40 - if you have any recommendations for changes to the default settings that this plugin installs I'd be happy to incorporate your suggested changes.

Cheers
Tim