metalman3797/Cura-Dremel-Printer-Plugin #91: "There are no file formats available to write with"

edgecase963, 2022-07-29:
I am currently using a Dremel 3D20 printer on Ultimaker version 5.1.0 (Linux Ubuntu).

It seems to load files, modify the models and slice them just fine. However, when I click the "Save to Removable" button to export the model, I get the error message "There are no file formats available to write with"

I've tried manually exporting it by going to File > Export, but it doesn't even give me the  ".g3drem" file format option.

timmehtimmeh, 2022-07-29:
@edgecase963 - thanks for the report.  It sounds like the plugin isn't being detected by Cura.  Does the plugin show up in the extensions menu?
[image]  If so please validate that the plugin is version 0.8.0, as prior versions are incompatible with Cura version 5.x

If it doesn't appear in the extensions menu then the plugin isn't installed.  If you recently updated to cura 5.x and haven't reinstalled the plugin you will need to install the plugin again.   Simply go to the cura marketplace and install it again.

I'll test the latest version of cura with ubuntu later tonight, or sometime this weekend.  If the plugin is currently installed, then it would be helpful if you could please send your cura.log, which can be found in:  ```$HOME/.local/share/cura/<Cura version>/cura.log```

timmehtimmeh, 2022-07-29:
@edgecase963 

I've run the following tests using Ubuntu 20.04.4 LTS:

1. First I wanted to simulate what would happen if a user was running Cura version 4.13 with Dremel Printer Plugin version 0.7.2 and then downloaded/ran Cura version 5.1.0.  The first thing that happens when a user launches Cura version 5.X is that they are presented with a message stating that the plugin can't be loaded (due to Ultimaker making breaking changes that required all plugins to update)
[image]

It's worth noting that at this step the printer plugin wasn't listed under the "extensions" menu.

2. From there I went to the marketplace and clicked on the gear icon (with the red notification on it) and clicked the "update" button on the dremel printer plugin.
[image]

3.  I then restarted cura, loaded benchy, sliced it and clicked "save to removable drive" and the file was successfully saved.
[image]

4.  Additionally Selecting `File->Export` has the option to export g3drem files
[image]

Please keep me posted with what you find.

Cheers
Tim

edgecase963, 2022-07-29:
At first, the plugin wasn't showing up in the Extensions drop down menu.

Oddly enough, updating the Dremel extension uninstalled it. Must have been a glitch because when I reinstalled it, *it worked!*
I am indeed using version 0.8.0 now and I'm able to save files in the .g3drem format.
Thank you for your help!

timmehtimmeh, 2022-07-29:
Glad that it was an easy fix! 🥳 

Cheers

closed 2022-07-29 by timmehtimmeh