metalman3797/Cura-Dremel-Printer-Plugin #94: Has this plugin been tested with other materials?

Fishhooks1945, 2022-10-17:
cura 5.1.1
dremel 3d45
PopOS! 22.04

I picked up a 3D45 and it already had a spool adapter so I can use esun spools off amazon. I am trying to print temp calibration towers with default PLA material setting, and it is only printing the base, and I'm wondering if it is related to this plugin. I have the toggle set the settings to follow the gcode no matter what. 

thank you for the help :)

timmehtimmeh, 2022-10-17:
@Fishhooks1945  

I wish it was a simple as a problem with the plugin, as then I could fix it.  Unfortunately temperature towers on the 3D45 seem to be a bit complicated.  I think that i have successfully printed one in the past, but it involved a lot of manual tweaking of the gcode, and unfortunately I can't recall what I tweaked to make the temperature tower work, and I tried again recently and was unsuccessful.

From my testing I verified that both the gcode and .g3drem file from cura both have the proper [M015](https://marlinfw.org/docs/gcode/M105.html) and [M109](https://marlinfw.org/docs/gcode/M109.html) commands, but unfortunately the printer never changed temperature when it was supposed to.  I think that the 3D45 firmware requires a very specific sequence in order for the temperature tower to work, and unfortunately I haven't been able to successfully find that sequence again.

FWIW the underlying gcode that the g3drem files use is the same gcode that Cura produces when it makes a normal gcode file.  The g3drem file just has a binary header in front of it that includes a picture and some additional information for the printer.

I'll keep this issue open, and if I find out how I was able to print the temperature tower I'll post here.  In the mean time if you do manage to successfully print the temperature tower then please post here what you did, as you're not the only person who's asked about the 3D45 and the temp tower.  In the mean time I'd encourage you to create a small bridge tower (i.e. one temperature section from the temperature tower) and print it at different temperatures.  

Cheers

Fishhooks1945, 2022-10-17:
Perhaps I will reach out to Dremel, or whoever they sold the rights to, to get a fix. I have previously been in contact with Greg Herker greg.herker@3pitech.com, he was the one that actually referred me to your Cura plugin. Maybe he will have some answers

I suppose I can just open the g3drem file with a hex editor or something and strip out the proprietary CRAP to verify the underlying gcode.. do you have a quick method for this, perhaps a python script that does it? I am very new to python, but this might be a good challenge for me. 

If I can't get this fixed, I'm just going to sell this Dremel to another unsuspecting user, and use the money to buy 3 brand new actually good printers. I like your suggestion of printing individual towers, but I shouldn't have to resort to time-consuming hacks to make this $1800 printer do what any $300 printer can do no problem.

timmehtimmeh, 2022-10-17:
>I suppose I can just open the g3drem file with a hex editor or something and strip out the proprietary CRAP to verify the underlying gcode.. do you have a quick method for this, perhaps a python script that does it? I am very new to python, but this might be a good challenge for me.

There's no need to do this - you can simply export a .gcode file from cura.  Also the gcode in a .g3drem file is in plaintext - you can simply open up a g3drem file in any text editor, edit the gcode section and print the new file.  

I sent this to someone else asking about the same thing:

If you haven't seen it yet, there's a Dremel Printer group on facebook [here](https://www.facebook.com/groups/882100391858795/)  with a bunch of helpful people.  This discussion is relevant to the temperature tower (and the pause at z that aborts after the specified number of layers): https://www.facebook.com/groups/882100391858795/posts/2918005671601580/  I know some of the engineers at Dremel monitor that site, and the last user I sent there may have posted something relevant since I sent him over.

metalman3797, 2022-10-17:
> think that i have successfully printed one in the past, but it involved a lot of manual tweaking of the gcode, and unfortunately I can't recall what I tweaked to make the temperature tower work, and I tried again recently and was unsuccessful.

Greg contacted me today about this.

On the 3D45 you need to enable “Prioritize G-code Settings”  Go to Tools > Settings > Adv. Mode and they need to check that “Prioritize G-code Settings”  is enabled under there.  It will work after that.

Fishhooks1945, 2022-10-17:
> 
> On the 3D45 you need to enable “Prioritize G-code Settings” Go to Tools > Settings > Adv. Mode and they need to check that “Prioritize G-code Settings” is enabled under there. It should work after that.

Yeah, I reached out to greg... I said in my email that I had that toggle enabled, I guess he missed that part

anyways, it doesn't work, even with that setting enabled.

Fishhooks1945, 2022-10-18:
for what it's worth, I removed the M117 line and it still stopped printing at the base. when I get home from school i'll remove the M104 S230 and see if the machine continues

Fishhooks1945, 2022-10-19:
I ran into another hangup, i've tried several USB drives and every time I plug one in, the Dremel LCD becomes unresponsive... so I cannot test edits to the gcode to diagnose the error. Every step with this machine, has been a challenge.... I am getting fustrated with it

timmehtimmeh, 2022-10-19:
@Fishhooks1945 -

I've had the 3D45 hang up when my text editor changed the text encoding of the file - the g3drem file needs to be encoded in UTF-8 format.  Some text editors change the encoding to something else when you save the file back out - for these experiments I'd recommend that you use .gcode (plain text) files and not .g3drem files.

Fishhooks1945, 2022-10-19:
That is a good tip. I think that is what is going on. I wasn't aware that I could strip all the proprietary stuff out and just run gcode, so I'll just do that and report back when i get out of school. Thank You!

timmehtimmeh, 2022-10-19:
You can also export gcode directly from Cura - just pick "save to disk" and select "gcode" from the dropdown

[image]

Fishhooks1945, 2022-10-20:
SUCCESS!!!! 

1) I didn't know that was an option in Cura to just save gcode, my OS shows the save file dialog a little differently, but indeed gcode is an option in the drop-down box
2) putting the gcode on a USB and loading it up worked fine. I have not yet tested the line endings thing, but i suspect you were right there. Pretty lame that the 3D45 just poops the bed. I will test this later though just so I know, but I probably won't use g3drem format ever again anyways
3) commenting out the [M117 code](https://marlinfw.org/docs/gcode/M117.html) allowed the towers to be printed. I only commented the first M117 out, and sure enough, when it got to the second one, the machine said "build completed" and stopped

But this is great news, I think I'll be able to print a full temp tower now, as long as I comment out the M117's. I'm not sure if this is something that can be done with the plugin, but my dad says he can show me a quick python script that adds a ';' in front of every M117, and I suppose I can add that to the post processing. Looking at the public FB group, it seems a lot of people run into this issue, so hopefully this helps others. 

Thank You for the help! I get paid with crypto when I do my chores, I'd love to send a thank you to you!

timmehtimmeh, 2022-10-20:
>SUCCESS!!!!

🚀 🎈🥳🎈 🥳 🎈🥳🎈🚀 
Awesome - thanks for sharing!  
🚀 🎈🥳🎈 🥳 🎈🥳🎈🚀 

----------------

>Pretty lame that the 3D45 just poops the bed

I suspect that one or two things is happening here, but the following is just a guess - 

Either 
1. When the [bitmap header](https://learn.microsoft.com/en-us/previous-versions//dd183376(v=vs.85)) gets converted to ascii the width and height and size change, and the printer tries to load a bitmap that is incorrect
2. The g3drem also has some binary offsets to the start of the bitmap, and start of the gcode, and same thing - when these get converted and saved out, the data changes and this causes issues.

As I said before, those are just guesses.  FWIW opening a g3drem and saving with VS code works fine, and on Win11, the updated notepad.exe will allow you to open the g3drem, but won't allow you to save, so the file type doesn't get converted.

---------------------------
>I probably won't use g3drem format ever again anyways

Totally up to you - the machine works equally well with both gcode and .g3drem files.  Personally I like having an image of the object that I'm printing that comes along with g3drem files.  

---------------------

>commenting out the [M117 code](https://marlinfw.org/docs/gcode/M117.html) allowed the towers to be printed. I only commented the first M117 out, and sure enough, when it got to the second one, the machine said "build completed" and stopped

I'm not sure where the M117s are coming from.  If I export a "normal" file using Cura and no post processing scripts then I don't get M117s in the file.  This might be part of whatever script is writing the temperature change commands  (I assume you're using the Change At Z post processing script).  

-------------

>I'm not sure if this is something that can be done with the plugin, but my dad says he can show me a quick python script that adds a ';' in front of every M117, and I suppose I can add that to the post processing.

 For now I'd recommend using the post processing portion of Cura as you mentioned.  It may be possible to ignore the command in the plugin, but I'm not sure right now.
[image]

--------------------------
>Looking at the public FB group, it seems a lot of people run into this issue, so hopefully this helps others.

>Thank You for the help! I get paid with crypto when I do my chores, I'd love to send a thank you to you!

That's very generous of you, but rather than ask for any form of payment I'd ask you for some of your time to post a complete writeup of how to print a temperature tower at the bottom of this thread so that the next time people ask they can get pointed to it. I'll link another closed issue to this one so that any users looking at that one will find your solution here.

Cheers
Tim

Fishhooks1945, 2022-10-20:
Thank you a ton for your help.. I am excited to be printing my temp tower! Your support has been excellent and clear! I'll consider using g3drem file in the future. I was able to load g3drem off USB when paying attention to not modify line endings (the basic text editor in my OS was being lame)

Since Dremel\3pitech people are in this thread, hopefully they can address some of these issues with the firmware to handle the errors gracefully, or at least give some indication of what's going on, like a timeout and error message

> I'm not sure where the M117s are coming from. If I export a "normal" file using Cura and no post processing scripts then I don't get M117s in the file. This might be part of whatever script is writing the temperature change commands (I assume you're using the Change At Z post processing script).

You are correct. I am using the AutoTowers plugin, which is an extension of 5axes's calibration shapes plugin, that handles the post processing and changing temp values. 5axes's plugin has in option in some of their scripts to turn off the M117's, but not the temptower one I was using haha. I tried using the 'search and replace' post processor, but it seems AutoTowers plugin runs last, so the search and replace didn't actually do anything. I'll just do this manually for now until the firmware is fixed. 

I will make a guide. It will take a minute, but I will get it done. I'm not sure if @timmehtimmeh is employed by 3pitech\dremel, but I'm not a huge fan of donating my time to companies selling very expensive printers for no compensation, versus open source projects... is the firmware to the machine open source? and if so, can you point me to where it is? 

THANK YOU again for your help, you've been fantastic and honestly the reason why this machine is still sitting next to my computer and not dumped onto someone else :)

timmehtimmeh, 2022-10-20:
>I will make a guide. It will take a minute, but I will get it done.

Thank you - and no rush - this will help a number of people who have struggled with this.  If you feel extra helpful you could make a bunch of temperature tower files, zip them up and attach them to the post...(i.e. file1 goes from 165 to 195...190 - 210 etc...

>5axes's plugin has in option in some of their scripts to turn off the M117's, but not the temptower one I was using haha.

It looks like [you've already commented](https://github.com/kartchnb/AutoTowersGenerator/issues/9#issuecomment-1284771602) on the autotowers plugin - 

I think that there is probably an easy way for this plugin to filter out M117 commands in this plugin when writing g3drem files.  The plugin gets the list of gcode from cura, and [writes each line of gcode individually](https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/blob/49f8345a8950ee9f0dd53d319e2750c24c2288d1/plugins/DremelPrinterPlugin/DremelPrinterPlugin.py#L726-L730). It should be relatively straightforward to add an "unsupported gcode" list that each individual line is compared against.  If the gcode to write is in the list then ignore it, and if it isn't then write it to the g3drem file.  This will then work for not writing M117s to g3drem files, but will not work when writing gcode files, as gcode files are written out by cura's [gcode writer plugin](https://github.com/Ultimaker/Cura/blob/main/plugins/GCodeWriter/GCodeWriter.py) (which incedentally is what this plugin is based off of).  Of course this will need time for testing before it could be released.

> I'm not sure if @timmehtimmeh is employed by 3pitech\dremel,

No, I'm not - when you install the plugin, or read the [Introduction section of the README](https://github.com/metalman3797/Cura-Dremel-Printer-Plugin#Introduction) you'll see that.

```  
The author of this software is not affiliated with Dremel, or Bosch Industries, or Ultimaker. 
Although the software functions well for the author and many other users, it is released without guarantee. 
Using the software and the files that it produces may damage your printer, set property on fire, 
or do other really_bad_things.
```

[image]

I wrote the plugin because the original software that came with my 3D20 didn't create supports, which severely limited what I could print.  That said I've found that each of the 3PI\Dremel team that I've interacted with have been extremely supportive & helpful.

>is the firmware to the machine open source? and if so, can you point me to where it is?

At the moment I don't think it is - I've spoken with some of the engineering team that worked on the printer and they are definitely pro-open source.  They were working behind the scenes to open source a bunch of their code, but corporate bureaucracy kept putting up roadblocks.  Maybe now that 3PItech has taken over things will be easier for them.

As an aside it is possible to [flash a 3D20](https://github.com/moonglow/FlashForge_Marlin/wiki/Installing-Marlin-on-Dremel-3D20) to the open source marlin firmware, however I'm not aware that anyone has managed to do the same for the 3D45.

>THANK YOU again for your help, you've been fantastic and honestly the reason why this machine is still sitting next to my computer and not dumped onto someone else :)

You are most welcome.  

As you've been successful I'm going to close this issue, but that shouldn't prevent you from adding an additional comment with  your writeup whenever you get time to make it.  Thank you again for volunteering to do that 👍 .  Cheers

closed 2022-10-20 by timmehtimmeh

Fishhooks1945, 2022-10-21:
>If you feel extra helpful you could make a bunch of temperature tower files, zip them up and attach them to the post...
The 5axes and AutoTowers plugins kinda do all this for you. 5axes provides the shapes and some post processors, but it requires some math and config to get a working gcode, whereas AutoTowers does it all for you based on your print settings. I think just a step by step on how to get good flow\temp\retract\etc towers without the machine crashing is all this needs to be, but let me know. 

>I think that there is probably an easy way for this plugin to filter out M117
hopefully they see this helpful suggestion :) I know they are having other random issues with M117, it seems to be a troublesome command that isn't exactly standardized between machines. In the meantime, I'll just manually scrub the M117's

>>I'm not sure if @timmehtimmeh is employed by 3pitech\dremel,
>No, I'm not

Sorry, I didn't really pay attention to that, I just noticed that this repo seems to be owned by Mike @ 3PItech. He emailed me a google document with a lot of helpful information and I'm feeling cautiously inspired to make a tkinter GUI with those curl commands, but I hope they open\free up some more AND support the community. I just don't want to put a bunch of work into something and feel taken advantage of, similar to how Fusion360 relies heavily on the community for support and tutorials, effectively subsidizing a multi-billion dollar company and locking users in with a prohibitave license (AND spools in this case). It would be a fun exercise for me, i've been looking for a reason to learn making a python GUI, but it has to be for the right reasons :-)
https://www.gnu.org/philosophy/open-source-misses-the-point.en.html
I hope I don't sound like a jerk, I've just been convinced this is the best path for my friends and I. I hope corporate makes the right decision. 

Thank You again, you have been a HUGE help in multiple ways, for issues not even caused by your work (that I know of). I made my first temp tower and it came out "meh", but now I will try the other AutoTowers calibration towers, and will report back with results and a guide :) Believe it or not I've spend many weeks staring at this machine and learning about 3D printing before I even tried printing something. I'm excited to get my esun spool dialed in!

timmehtimmeh, 2022-10-21:
>I just noticed that this repo seems to be owned by Mike @ 3PItech

This wasn't true when you opened the issue this past Monday.   If you look at the readme you'll see links to my github repo all over it.  Also if you go to https://github.com/timmehtimmeh/Cura-Dremel-Printer-Plugin you'll see that it forwards to this repo.

I'm currently in the process of transferring the plugin to them, as the 3pi tech engineers have expressed interest in taking over development of the plugin (I've maintained it alone in my free time for years).  I've had to semi-reverse-engineer the g3drem format, and do a lot of development by trial and error, and I currently don't have a lot of free time to develop new features or maintain the plugin/support the plugin.  The engineers at 3pitech are excited to develop it, and getting paid to do so.  The plugin will remain open source (as GPL3 demands) but will hopefully get more features than I can implement as the engineers have access to knowledge about the firmware that I don't (and don't have time to work on even if I did)

>hopefully they see this helpful suggestion :) I know they are having other random issues with M117, it seems to be a troublesome command that isn't exactly standardized between machines. In the meantime, I'll just manually scrub the M117's

I've already implemented a fix for this, and will be making a pull request on what is now no longer my repo.

Fishhooks1945, 2022-10-21:
wow, that is awesome, and you are fast! :+1: I hope to be as good some day. I'll have to do some reading on how I can test code because I want to try! 
I was able to print a flow tower with no crashing, but the retraction tower bugged out the machine. I know this is more of an issue with the dremel firmware now, but is it OK to post the results in a new issue on this repo? 
Thank you again. I hope I have not offended you with the ownership thing.

timmehtimmeh, 2022-10-21:
After a bit more testing this morning I found an issue with the pull request and I closed it.  I have always thought that the items in the `gcode_list` were individual gcode lines, but after testing it seems that each item in the `gcode_list` contains all the gcode for one layer.  Therefore the code in the pull request was removing all the gcode for any layer that contained a `M117`.  The autotowers plugin makes it a bit more difficult to detect this, as repeating the same test twice creates different gcode each time, so searching through diffs hid the issue behind a bunch of other changes.  I think I've now fixed it for real on my branch, but I want to test it this weekend before I submit another pull request.  

>I was able to print a flow tower with no crashing, but the retraction tower bugged out the machine. I know this is more of an issue with the dremel firmware now, but is it OK to post the results in a new issue on this repo?

It's not my repo anymore, and as I said I'll be stepping down from maintaining it, but I'd say please feel free to create issues.

>I hope I have not offended you with the ownership thing.

Not at all - I'm hard to offend - I understand where you're coming from, but I'd hope that you at least give Mike and the engineers a chance - the fact that the engineering team is emailing you documents and information should at least prove to you that they're willing to work with the community.

timmehtimmeh, 2022-10-22:
@Fishhooks1945 

When you were printing the temperature towers did you actually see the temperatures change?

Fishhooks1945, 2022-10-22:
@timmehtimmeh 
yes, I did. I didn't see the M117 lcd messages obviously, but I did see the circular temperature gauge thing change accordingly. That was cool :)

metalman3797, 2022-12-12:
> perature towers did you actually see the temperatures change?

There is a new firmware under development that is going to simplify tissues like this and allow much more control over temps, fans, print speeds etc :)