metalman3797/Cura-Dremel-Printer-Plugin #100: Plugin DremelPrinterPlugin was not loaded because it could not be verified.

plgunn, 2022-12-01:
I get the above error message when I start Ultimaker Cura Enterprise 4.13.2.  Although the plug-in is shown as installed within the marketplace, when I check the box of the plug-in and restart Cura, the error message is displayed.  The machine is running Windows 10 Enterprise.

timmehtimmeh, 2022-12-02:
@plgunn -

Which version of the plugin is shown as installed?

I don't have much knowledge of Cura Enterprise, but I wouldn't be surprised if Ultimaker blocked this plugin for Cura Enterprise, as they don't have any dremel printers, and have no way of really testing it.  I've certainly never gotten any communication from Ultimaker that this plugin has gone through their validation process, and I've never applied for it, so the fact that it's listed in their marketplace in the Cura Enterprise edition is news to me.

[image]

Is it possible that the computer had a non-enterprise version of Cura on it with this plugin installed at some point before?

plgunn, 2022-12-02:
Thanks, Tim.  The version of the plug-in is 0.7.2.  

I'm using the 4.13.2 version of Cura because that is the only version that I've found supports the Postprocessing Script that allows the print to be paused at a specified layer.  I also have the 5.2.1 version of Cura installed and I have the Dremel Printer plug-in installed and working with this version.  However, the Postprocessing Script I found to pause at a layer doesn't appear to support the 5.2.1 version of Cura.

closed 2022-12-02 by plgunn

timmehtimmeh, 2022-12-02:
Thanks for the reply with additional info.  As far as I can tell, the plugin being blocked is a function of Cura Enterprise - if it's blocking issue for you and you don't need the enterprise additionse I'd recommend trying normal (non enterprise) [Cura 4.13.1](https://github.com/Ultimaker/Cura/releases/download/4.13.1/Ultimaker_Cura-4.13.1-amd64.exe).  This plugin won't show up oin that version of Cura's marketplace, but you can download the plugin version 0.7.2 from [here](https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/releases/download/0.7.2/Cura-Dremel-Plugin-0.7.2.curapackage), launch (non enterprise) Cura 4.13.1 and drag and drop the .curapackage file onto it to install the plugin.

Cheers

plgunn, 2022-12-02:
Thanks, Tim.  Sounds like a solution worth trying.