metalman3797/Cura-Dremel-Printer-Plugin #104: 3D45 Does not preheat before starting

e404nothing, 2023-12-16:
I am using a Dremel 3D45 and after slicing the object with the plugin in Cura, and then exporting as both .g3drem and .gcode. When I preview the files in the printer, there is an image for the .g3drem file but not the .gcode file as expected. But when I select either file, it doesn't tell me what either temperature its going to preheat to. Furthermore, when I begin printing them, the printer parks the head in the front left as if it were to pre-heat, then slowly moves into position and begins printing incredibly slow.

timmehtimmeh, 2023-12-16:
@e404nothing - gcode won't have a preview image in it - that's expected - one of the features that the proprietary g3drem file format adds is the preview image.

Regarding the slow printing - the gcode is entirely produced by Cura, so I'd start by looking at the speed values

[image]

What version of Cura are you using? 

Although I no longer control this repo and cant' make any changes, i can say that I haven't seen this behavior on my 3D45.  It'd be helpful if you are able to upload a .g3drem file that was sliced with your settings (ideally something small like the [calibration cube](https://www.thingiverse.com/thing:1278865))

e404nothing, 2023-12-16:
Thanks for the reply, I quickly sliced up a calibration cube and tested the print with that. The slow movement didn't happen this time, it might have only came with the .gcode file or I was overanalyzing the speed as I am also trying to print TPU (slowly). Below is a snippet of the initial bit of code, I am using VSCode with the .txt interpreter. I understand the initial code is ASSIC or something but I am not familiar with that language, any insight on that is helpful. I was unable to copy the binary after the "g3drem 1.0      :" so I copied after it as well. If there is a way to upload the file I am all ears, I tried uploading it as an attachment but it said it wasnt a supported file type. 

As for what happed when I ran this code; the picture preview came up, it did not show preheat temperatures before starting print, when initialized it homed, and  then went to the "preheat park location" (front left) and then after pausing for a second it moved to the centerish to begin printing all the while being at room temperature. 

g3drem 1.0      :
;Cura-Dremel-Printer-Plugin version 0.8.1
;Printing on: Dremel3D45
;Using material: "TPU 95A"
;Quality: "High Speed"
;FLAVOR:Marlin
;TIME:1000
;Filament used: 1.39334m
;Layer height: 0.34
;MINX:-11.017
;MINY:-45.951
;MINZ:0.3
;MAXX:16.983
;MAXY:-17.951
;MAXZ:20.02
;Generated with Cura_SteamEngine 5.2.1
M104 S228
M105
M109 S228
M82 ;absolute extrusion mode
G28
G1 Z50.00 F400
G92 E0
G1 F200 E3
G92 E0
M132 X Y Z A
M907 X100 Y100 Z50 A100
G92 E0
G92 E0
G1 F6000 E-6.5
;LAYER_COUNT:59
;LAYER:0
M107
M204 S1200
M205 X8 Y8
G0 F3600 X-6.362 Y-17.951 Z0.3
M204 S800
;TYPE:SKIRT
G1 F6000 E0
G1 F1500 X-7.06 Y-17.96 E0.03483
G1 X-7.623 Y-18.008 E0.06302
G1 X-8.174 Y-18.134 E0.09122

timmehtimmeh, 2023-12-16:
@e404nothing  - thanks for the information

Unfortunately I don't have a ton of time to investigate this right now, but I sliced up the calibration cube with PLA settings and am currently printing it.

The printer started out by calibrating the build plate, then preheating as expected:
[image]

After about four to five minutes the printer was almost up to temperature:
[image]

Afterwards the print started and the speed was normal
[image]

Here's the file I used for my test
[image]

Download this g3drem file and try printing it with PLA - if that works then we'll know that there's some difference in your slice settings and/or cura version (looks like you're running Cura 5.2.1).  If it also exhibits issues then my guess is that there's something wrong with your printer, as the file is working on my 3D45.

[D_xyzCalibration_cube.zip](https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/files/13693243/D_xyzCalibration_cube.zip)

FWIW you can zip up files and upload them =)

Cheers

e404nothing, 2023-12-16:
@timmehtimmeh Thanks that was super helpful!
Turns out I needed to reset my firmware, after I did that it worked immediately.

timmehtimmeh, 2023-12-16:
@e404nothing - happy to help!

Thank you for posting what you did to solve it - hopefully that'll help someone else in the future if they come across the same issue!

closed 2023-12-16 by timmehtimmeh