metalman3797/Cura-Dremel-Printer-Plugin #105: Typo in Package.json (Plugin version 1.0.0 won't install on Cura 5.6.0)

timmehtimmeh, 2024-03-26:
I just tried manually installing the 1.0.0 release of the plugin by dragging and dropping it onto my existing version of Cura 5.6.0 which was running the plugin version 0.8.1.  Cura properly gave me the "the plugin will be installed when cura restarts" message however when I re-launched cura the plugin version did not update to version 1.0.0

I then deleted the cura directory where the plugin gets installed (on windows it's %APPDATA%/cura/5.6) and re-launched cura.  I then dragged and dropped the 1.0.0 plugin onto cura, and again received the message to restart again
[image]

Upon restart and navigating to the "select non ultimaker printer" the "Dremel" option was not present
[image]

Adding an Ultimaker S5 and checking the Extensions menu I noticed that there is no Dremel printer plugin option
[image]

Investigating this a bit further I looked at the cura.log and noticed these lines
```
2024-03-26 15:50:30,204 - DEBUG - [MainThread] UM.PackageManager.getPackageInfo [678]: Found potential package.json file 'package.json'
2024-03-26 15:50:30,204 - ERROR - [MainThread] UM.PackageManager.getPackageInfo [689]: Exception: Failed to load potential package.json file 'package.json' as text file.
2024-03-26 15:50:30,205 - ERROR - [MainThread] UM.PackageManager.getPackageInfo [689]: Traceback (most recent call last):
2024-03-26 15:50:30,206 - ERROR - [MainThread] UM.PackageManager.getPackageInfo [689]:   File "UM\PackageManager.py", line 681, in getPackageInfo
2024-03-26 15:50:30,206 - ERROR - [MainThread] UM.PackageManager.getPackageInfo [689]:     package_json = json.loads(f.read().decode("utf-8"))
2024-03-26 15:50:30,206 - ERROR - [MainThread] UM.PackageManager.getPackageInfo [689]:   File "json\__init__.py", line 346, in loads
2024-03-26 15:50:30,207 - ERROR - [MainThread] UM.PackageManager.getPackageInfo [689]:   File "json\decoder.py", line 337, in decode
2024-03-26 15:50:30,207 - ERROR - [MainThread] UM.PackageManager.getPackageInfo [689]:   File "json\decoder.py", line 353, in raw_decode
```

I then opened up the pacakge.json included in the plugin and saw that the comma on the `"package_version":"1.0.0"` was inside the quotation mark [original file here](https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/blob/113f601ca8c7888ab1dfe02ba11f0061661f8458/resources/package.json#L11)

    ```
"package_id": "DremelPrinterPlugin",
 "package_type": "plugin",
 "package_version": "1.0.0,"
 "sdk_version": 8,
```

timmehtimmeh, 2024-03-26:
Upon fixing the json, re-zipping the file, and repeating the process outlined above (clear out the cura version directory, reinstall the plugin) I got further.  

```
2024-03-26 16:08:08,567 - DEBUG - [MainThread] UM.PackageManager.getPackageInfo [678]: Found potential package.json file 'Cura-Dremel-Plugin-1.0.0/package.json'
2024-03-26 16:08:08,568 - INFO - [MainThread] UM.PackageManager.installPackage [515]: Package [DremelPrinterPlugin] version [1.0.0] is scheduled to be installed.
```

Then after cura restarts:
```
2024-03-26 16:08:16,813 - INFO - [MainThread] UM.PackageManager._installPackage [607]: Installing package [DremelPrinterPlugin] from file [C:\Users\Tim\AppData\Local\cura\5.6\cache\cura_packages\DremelPrinterPlugin.curapackage]
```

however immediately afterwards there is another set of warnings:
```
2024-03-26 16:08:16,813 - WARNING - [MainThread] UM.PackageManager._installPackage [643]: The path C:\Users\TIM\AppData\Local\Temp\tmp8xehcra1\files\plugins does not exist, so not installing the files
2024-03-26 16:08:16,813 - WARNING - [MainThread] UM.PackageManager._installPackage [643]: The path C:\Users\TIM\AppData\Local\Temp\tmp8xehcra1\files\materials does not exist, so not installing the files
2024-03-26 16:08:16,813 - WARNING - [MainThread] UM.PackageManager._installPackage [643]: The path C:\Users\TIM\AppData\Local\Temp\tmp8xehcra1\files\qualities does not exist, so not installing the files
2024-03-26 16:08:16,813 - WARNING - [MainThread] UM.PackageManager._installPackage [643]: The path C:\Users\TIM\AppData\Local\Temp\tmp8xehcra1\files\variants does not exist, so not installing the files
2024-03-26 16:08:16,814 - WARNING - [MainThread] UM.PackageManager._installPackage [643]: The path C:\Users\TIM\AppData\Local\Temp\tmp8xehcra1\files\images does not exist, so not installing the files
2024-03-26 16:08:16,814 - WARNING - [MainThread] UM.PackageManager._installPackage [643]: The path C:\Users\TIM\AppData\Local\Temp\tmp8xehcra1\files\intents does not exist, so not installing the files
2024-03-26 16:08:16,814 - WARNING - [MainThread] UM.PackageManager._installPackage [643]: The path C:\Users\TIM\AppData\Local\Temp\tmp8xehcra1\files\intent does not exist, so not installing the files
```
and the plugin does not appear in the set of extensions.  I'll continue investigating and post anything I find as a comment below

timmehtimmeh, 2024-03-27:
The warnings listed in my comment above are not the cause of the issue.  Even when I follow the same steps but install the 0.8.1 plugin the same lines appear.

It does appear that the DremelPrinterPlugin class is defined a second time, and the second definition is causing some issues:

It's first defined [here](https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/blob/113f601ca8c7888ab1dfe02ba11f0061661f8458/plugins/DremelPrinterPlugin/DremelPrinterPlugin.py#L58): 
`class DremelPrinterPlugin(QObject, MeshWriter, Extension):`

and further down the code is also defined [here](https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/blob/113f601ca8c7888ab1dfe02ba11f0061661f8458/plugins/DremelPrinterPlugin/DremelPrinterPlugin.py#L316):
`class DremelPrinterPlugin:  # Assuming this is part of a class definition`

removing [these lines](https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/blob/113f601ca8c7888ab1dfe02ba11f0061661f8458/plugins/DremelPrinterPlugin/DremelPrinterPlugin.py#L313-L316), 
```
import os
from UM.Logger import Logger

class DremelPrinterPlugin:  # Assuming this is part of a class definition
```
fixing the package.json file as outlined in the initial comment, and re-calling the make_release script has finally allowed me to install plugin version 1.0.0 🥳 

[image]

GINNOV, 2024-03-27:
I don't see the 1.0.0 in the CURA marketplace, the only way to install it is via the manual process? thx

timmehtimmeh, 2024-03-27:
@GINNOV 

The ultimaker team has a process to review each plugin after it's submitted to validate that it doesn't break anything and that it's safe for cura users.  Historically this took a few days.  

As it stands now the manual install won't work either due to the bugs I outline in this bug report.  This should be a relatively quick fix for them to implement and re-release.

(@metalman3797  - ping me if you have any questions)

In this particular case my guess is that the current plugin submission (I'm assuming they have submitted it) will be rejected, as the currently released version 1.0.0 doesn't actually install due to the bugs I outlined above.  One of the two bugs (editing the package.json) is only needed for people who want to install the plugin manually, and the ultimaker servers generate that file on their own based on the information submitted when you upload a new version of the plugin for ultimaker to review.  

Assuming it's not just my computer experiencing the error where the plugin doesn't install, then Mike and his team will have to fix the error and re-submit before the new version is accepted to the marketplace.  This is Mike's first plugin update since I handed the repo over to him, so he's a bit unfamiliar with the process.  As the former owner/creator of the plugin I'll ask for a bit of patience for the 3pitech team as they figure  the process out.

GINNOV, 2024-03-29:
thank you! Appreciate the details