metalman3797/Cura-Dremel-Printer-Plugin #107: Plugin version 1.0.0: Plugin preferences no longer pops up preference dialog box

timmehtimmeh, 2024-03-27:
In version 0.8.1 navigating to extensions->dremel printer plugin->preferences
[image]

Would pop up a dialog box allowing the user to configure a few settings:
[image]

However clicking the the 1.0.0 release the dialog box no longer pops up,
[image]

timmehtimmeh, 2024-03-27:
The issue seems to be related to the lines:

`color: "#000000"` in the DremelPluginprefs.qml file.  Removing the three lines that specify the black color allows the popup to appear:

[image]

The lines I removed are:
[here](https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/blob/113f601ca8c7888ab1dfe02ba11f0061661f8458/plugins/DremelPrinterPlugin/DremelPluginprefs.qml#L48)
[here ](https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/blob/113f601ca8c7888ab1dfe02ba11f0061661f8458/plugins/DremelPrinterPlugin/DremelPluginprefs.qml#L63)and
[here](https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/blob/113f601ca8c7888ab1dfe02ba11f0061661f8458/plugins/DremelPrinterPlugin/DremelPluginprefs.qml#L104)

Sidenote - the window should be made a bit bigger by changing the [minimumHeight ](https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/blob/113f601ca8c7888ab1dfe02ba11f0061661f8458/plugins/DremelPrinterPlugin/DremelPluginprefs.qml#L15)to properly contain the new IP address element.  When the window first pops up it's a bit smaller than the items it contains:
[image]

metalman3797, 2024-05-01:
Approved changes.  New build uploading today.

closed 2024-05-01 by metalman3797