metalman3797/Cura-Dremel-Printer-Plugin #108: Problematic working on Cura 5.7 & Minimum speed is needed

daUser, 2024-06-13:
Hi,

this PlugIn works for me on Cura 5.7.1 under Windows, but not in the Docker Variant linuxserver/cura:5.7.2 or : 5.7.1

I have openend a issue on the cura-Projekt an they say maybe the plugin needs an update for the current cura version: https://github.com/Ultimaker/Cura/issues/19222

Can you do anything?

And, when you work on it: Cura needs now a minimum speed for all Materials >0: https://github.com/Ultimaker/Cura/issues/15332#issuecomment-1520855904

Maybe you can change this also?

daUser, 2024-06-13:
Maybe i found the error:
The cura Martkplace installed Version 0.8.1. Now i have to look how to install manually.

timmehtimmeh, 2024-06-13:
@daUser - I can't comment on the the linux/docker issue that you're having, but the minimum speed bug was already filed https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/issues/102 and is fixed in the [1.0.0 release](https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/releases/tag/1.0.0) (see release notes).

**Note** that the 1.0.0 release had some problems that I raised and fixes were merged in on the 28th of March (see https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/issues/105), however those fixes are not reflected in the 1.0.0 release that you can download from the link above.  If you try to install the plugin from the release at the link above the plugin will not install.  

For your convenience I've made a fixed copy of the release and attached it here: [Cura-Dremel-Plugin-1.0.0.zip](https://github.com/user-attachments/files/15823466/Cura-Dremel-Plugin-1.0.0.zip)  I'm not sure if manually installing will work with docker, but with a normal setup to install it you would need to do the following:

1. Unzip the Cura-Dremel-Plugin-1.0.0.zip
2. open cura
3. drag and drop the extracted Cura-Dremel-Plugin-1.0.0.curapackage onto the open cura window. 
4. Close and restart Cura when prompted [image]
5. Validate that the 1.0.0 plugin is installed by opening Extensions->Dremel Printer Plugin->and look for the version number
[image]

I'm also not sure on the status of getting the 1.0.0 version officially released on the cura marketplace - @metalman3797 controls that now, so maybe he can comment on it.

MC68030, 2024-08-26:
Within Cura 5.8, i have the same issue , Dreemel 3D20(native installation, OS-X).
On the market place, it's still version .0.81 that will install. So i followed this thread, and upgraded manually to 1.0.0, which is confirmed at the Plugin Page.

It turn's out, it's not related to this version, it's related to the Dremel PLA i selected.Since i still have some of the original 0,5 kg spools, i selected this PLA. As soon as i switched to the 0,75 kg Dremel PLA, the error was gone.

But that's not really a solution for me, since the definitions in terms of temperature etc. are differnt between those two

timmehtimmeh, 2024-08-26:
@MC68030 - thanks for the info. Apparently the minimum speed setting got missed for the old 0.5kg spools.

There are two ways you can fix this until it's fixed officially in the next release.  (**note** I don't control the releases anymore, so I can't give you any estimate on the timeline for that), Below is the recommended one

1)  Select the 0.5kg filament, and the quality setting you want to print with, then click the "show custom" box
[image]
2) from there type "speed" (without the quotes) in the search window and scroll down to cooling
[image]
3) Under the minimum speed setting (that should be highlighted in red, type in 10mm/s
4) Save the setting to a new custom setting by clicking on the floppy disc icon in the print settings
[image]
selecting save as new profile

[image]

And pick an appropritate name (in my case I had selected the "high quality" profile, so this one will be named "high quality with min speed"
[image]

Then any time you want to print with these particular settings, just select them from the quality profile
[image]

The other way is to edit a text file that's installed when the Dremel plugin gets installed, but it's more complicated, and in some situations the edits you make to the file could be overwritten, so I wouldn't recommend it unless the above doesn't work.

MC68030, 2024-08-26:
Thank you for the very fast response! Indeed, that fixed it for me, and i applied this also to the other two profiles.