metalman3797/Cura-Dremel-Printer-Plugin #109: Crashes when I try to add Dremel 3D40

TheJakeZilla, 2024-07-05:
When I try to add the Dremel 3D40 Cura Crashes

https://github.com/metalman3797/Cura-Dremel-Printer-Plugin/assets/105027956/13e005c7-93ab-42b8-8d7d-941afb5f59c3

timmehtimmeh, 2024-07-05:
@TheJakeZilla - I see that you're running Cura 5.4.0 -  A few questions:

1. What version of the plugin are you running?   You can find the version by opening the Extensions menu and selecting the dremel printer plugin -
[image]
2. Does the plugin crash when you try adding the 3D20 or the 3D45?

I cant look at this immediately, so I'll kindly ask for your patience - also I might need you to get a cura.log file so that I can further debug, but not yet...

TheJakeZilla, 2024-07-09:
I am running plug-in 0.8.1 (This is the version I got off the Cura Menu the other day) and cure version 5.4.0

It does not crash when adding the

Attached is the Cura.Log.1 file, I had to save it as txt to post it. I'm not sure if this log contains the crash because I reopened Cura to test the other printers. 
[curaLog.TXT](https://github.com/user-attachments/files/16150427/curaLog.TXT)
 other two printer types. 

Don't worry about looking at this Immediately, We are currently using Dreamlabs and it works fine, but I would love access to the QOL improvements of Cura without transferring programs.

timmehtimmeh, 2024-07-09:
@TheJakeZilla  - Thanks for the log - looking at it it appears that something has corrupted the installation of the plugin, as I don't see the standard message in it saying that Cura has successfully loaded the DremelPrinterPlugin, and I don't see any of the expected log messages from the plugin indicating that the plugin is initializing.   One of the first things that the plugin does when it's loaded is check to ensure that all the plugin files are installed, and if not then it re-installs them.  I don't see that happening in your logs, so something isn't working as expected in your case.

FWIW - I just tried to duplicate what you were seeing by installing Cura 5.4.0 on a windows 10 machine and downloading the plugin version 0.8.1 from the marketplace, and restarting Cura.  When I restarted I was able to successfully load the 3D40 with no crash or other problems.

My cura.log (located in %APPDATA%\cura\5.4\cura.log) shows that the plugin is called by cura and installs the printer definitions etc...
[image]

and then Cura indicates that the plugin is loaded
[image]

however when I search your cura.log file for the text `Loaded plugin DremelPrinterPlugin` I don't find that text anywhere, indicating that in your case Cura hasn't properly loaded the plugin.  I'm unsure what's going on exactly, but the first thing I'd try is to delete the plugin and reinstall it.  

To do so
1. Select the Marketplace and search for dremel and uninstall the plugin
3. When prompted close cura
4. Navigate to `%APPDATA%\cura\5.4\` and delete the following items
-  cura.log
-  plugins\DremelPrinterPlugin (delete the folder if it exists)
[image]

5. Restart Cura
6. Validate that the dremel printer plugin is uninstalled by checking the extensions menu and confirming that there is no Dremel Printer Plugin listed
[image]

7. Reinstall the plugin from the marketplace
8. When prompted, quit & restart Cura
9. Try re-adding the 3D40 - if that still crashes then please upload the newly created cura.log (at %APPDATA%\cura\5.4\cura.log) here.